wokwi/wokwi-features #526: Missing examples in code about measuring code running time (bug) #525 #526

kvapehe, 2023-02-16:
**Duplicate of issue #525**
which may now be closed / removed. Reason for this duplicate is that the other issue was not labeled Bug

**Describe the bug**
Missing code examples on the page

These lines failed to work (and more)
https://wokwi.com/share/embed/sphP74qbXQziV509bWCp
https://wokwi.com/share/embed/eroVOJ8X1jcNY6TKBKiu
And this worked: https://wokwi.com/share/embed/282702715864220173

**To Reproduce**
Open the page follow the links and look on the details on this issue-page.
https://blog.wokwi.com/how-to-measure-the-speed-of-arduino-code/

**Expected behavior**
A blog that is fully working, as it may have worked in June 2020, and now after some possibly reorganisation of the wokwi site
is not working any more. The Bug may affect other blog articles with same iframe patterns as referred to in this issue Bug report

**Environment (please complete the following information):**
Not relevant, as the situation happens in this blog article: (and others??)
https://blog.wokwi.com/how-to-measure-the-speed-of-arduino-code/

**Additional context**
Now following all details. This issue is restructured to fit the Bug report structure..
Some links are mentioned more than once to "catch interest" or be shown in the first headings.
Then mentioned later to give more valuable detailed information if needed.

***Details of Bug Report***
Missing examples in code about measuring code running time (bug) #525 
https://blog.wokwi.com/how-to-measure-the-speed-of-arduino-code/

Issues reading a blog on your blog page.
I dug into the code referenced in the iframes and found this pattern.

These lines failed to work (and more)
https://wokwi.com/share/embed/sphP74qbXQziV509bWCp
https://wokwi.com/share/embed/eroVOJ8X1jcNY6TKBKiu
https://wokwi.com/share/embed/d8PkzYWfzDI0GcI3UJk5

while these lines still worked
https://wokwi.com/share/embed/282702715864220173
........._https://wokwi.com/projects/282702715864220173
........._https://wokwi.com/projects/282701458229428749

I think the author of wokwi may be the only one to fix all missing iframes (if possible) in the article on
https://blog.wokwi.com. It is somewhat possible to see which lines failed due to the "code" after the last /
in all the links above.

The request is that you read the article and investigate what has happened.
I can see the article is written in June 2020 and assume the locations for source code has changed.,
and that has damaged the article posted.

I would expect this Bug or issue for blog articles may affect more than this article when the pattern is as described above.
Potentially other affected articles: When author is other than wokwi site owner, code is implemented with iframes and article written in 2020 or earlier.

Let me know if any more details are required to investigate this issue.

Regards

urish, 2023-02-16:
Thanks for reporting, fixed!

closed 2023-02-16 by urish

urish, 2023-02-16:
Some more details about why this happened:

Very early in the history of Wokwi, we used another database, which used a different scheme for project ids (like "sphP74qbXQziV509bWCp" and "eroVOJ8X1jcNY6TKBKiu"). We only had 181 projects in that historical database, and we kept supporting it. A few months ago, we removed support for the old database, and tried to locate all the links still pointing to these very old projects and updating them. Apparently we missed a few. If you find any more places where we still have links to those old projects, please let us know!

kvapehe, 2023-02-16:
Great support.
I figured it was something like that. It seems to me only you can have a overview of all projects.
Since this was from 2020 then maybe a search for that helps. I will report when I find, duirng my own fail and try different stuff.
I could see a pattern here, thats whay I was "sure" you could fix it.

What I do miss though is a way to see all the projects on wokwi.
Until now I use one browser not logged on and see all examples on the page, then read all blogs and docs to look at the links in those plcess, then I save projects I like an make my own inventory.

Maybe it is alrady on your Roadmap so this is not a request, I just miss a way to find info myself, also when digging.
Until now I have just let the "luck" show me cool projects and then share own or known projects to the public so som may or may not accidently find them later.

Anyway. Nice work again.

kvapehe, 2023-02-16:
Especcially I like the detailed answer from you. I guess you have showed information valid to more than me with that post.
I just happen to have this ability to if something can and may go wrong, it will eventually happen (read always happen) when I am involved. Now it was about timing and rgisters I barely understand, and then som not working examples, but I almost knew the content anyway. Living and learing is a great motto.