wokwi/wokwi-features #527: Norwegian translations

kvapehe, 2023-02-16:
**Norwegian translations**
Post last updated: 2023-02-16
Translation last updated along this post: 2023-02-11
This issue is not labeled Translation as of 2023-02-16

This issue, which may be labeled translation is created to better facilitate updates of 
the Norwegian Bokmål translated text.
Link to wokwi: https://wokwi.com/projects/new?lang=nb-NO

One immediate change would be to change back from Språk which in Norwegian means Language. (menu account)
This may help other than Norwegians to change back to another language if they should end up seeing the Norwegian translation.

The already submitted json-file from issue: #221is attaced. (Note read the section for advices updating or creating your own version)
In the #221 issue the translation was sent as a comment instead of making this separate issue #527
Link to original translation https://github.com/wokwi/wokwi-features/issues/221

**Link to original English translation**
https://wokwi.com/api/i18n/en.json

**Link to Norwegian translation already submitted and implemented**
https://wokwi.com/api/i18n/nb-NO.json

**Issues regarding the translation**
In US English capitalisation is often used for separate words. In UK-English lower case is used for all but the first word.
In German all substantives are capitalised. In Norwegian very few words are capitalised, even the first word unless the Norwegian grammar requires it. One example. Norwegians, Norwegian is translated to "norske med norsk språk" Only the word Norway is capitalised when using the Norwegian grammar. 
Knowing this often "complicates" when SW is translated as you need to weight the grammar vs the appeareance in the program you are translating.

If anyone find issues You may either download the English translation file and start over again and send it as a link to this issue,
or you may save yourself some time by fixing the Norweigan Bokmål translation or use the Norwegian Bokmål translation
as a start/template for the Nynorsk translation if you feel for it.

**Advice**
Simply ask the author of wokwi if any help is need and help out making this great project available to a bigger Norwegian audience.

**Further ahead**
The plan is to revisit the Norwegian translation some more times in the near future, to improve it further.

**Considerations**
The translation was somewhat freely written and not completely accurate to the English text.
The idea is to include words the Norwegian users would possibly use and sometimes an English word is better than a
Norwegian translation. I.e. LED, TFT, LCD All may be translated to Norwegian like a LED is Lysdiode, but the words chosen are there to help the users find the components. When possible both languages may have been used.

Another is Log on, Logg Inn, Logg inn, Logginn and so on. Different SW uses all the wordings mentioned, but one has been chosen for now.

**Submitted data**
Note: Do not use this. Chose instead the link above, than save that json-file to your computer to get the newest version
json-file submitted, is attached to this issue:
The file is included as a reference:
[nb-NO_json.txt](https://github.com/wokwi/wokwi-features/files/10752351/nb-NO_json.txt)

Another note: as json is not allowed as attachment on this site, either name the file .txt to see it online, or put it inside a .zip-file.
The method chosen 2023-02-11 was a .txt-file.

urish, 2023-02-19:
Thanks! Looking forward for any future updates in the Norwegian texts.

kvapehe, 2023-03-18:
A new translation will follow soon.
Some elements have been changed

kvapehe, 2023-03-18:
2023-03-19 Added new/updated translated file for Norwegian Bokmål

Suggested naming in Menu: Norsk bokmål

[nb-NO_.json_2023_03_19.txt](https://github.com/wokwi/wokwi-features/files/11009824/nb-NO_.json_2023_03_19.txt)