wokwi/wokwi-features #529: Bug or error in my code?

FernandoGarcia, 2023-02-17:
Hi!

I have have found a strange behaviour in [this simulation](https://wokwi.com/projects/356550855830648833) but I'm not sure it's my code or the simulator since I don't have experience with stepper motors.

After move the switches to right side and press the green button the motor should spin in CCW direction because the relative position to move on code is `-10000` but the motor spin to CW direction and after release the button and press again the direction can change to CW as you can see is this video at 0:24 and steps ~500.

https://user-images.githubusercontent.com/2778501/219651647-0ba26f2a-794a-4562-b583-0db38c967c2f.mp4

Moreover the red button makes the motor spin to CW direction as expected when in the line 68 on setup I have this value `10000` but  the behaviour of red button changes when the value is `-10000`

Thanks for your time.

Best regards.

Dlloydev, 2023-04-02:
Seems to be an issue with the library. Perhaps this is a workaround:
```c++
    if ((debouncedState[0] == LOW) && (button[0].currentDuration() > 200))
    {
      lcd.setCursor(0, 0);
      lcd.print("Moving CCW      ");
      myStepper.setCurrentPosition(myStepper.currentPosition());
      myStepper.move(-10000);
      myStepper.run();
      delay(5);
    }
```
Try your code using a different library [here](https://wokwi.com/projects/360882572386122753).

closed 2023-04-02 by urish

FernandoGarcia, 2023-04-02:
Hi @Dlloydev !

I think the workaround won't solve the problem efficiently because it slow down the rotation speed.

Anyway thanks for take time to debug my code and show alternative solution.

Best regards.