wokwi/wokwi-features #532: Managing My projects

alto777, 2023-02-18:
I was astonished to see that I have > 500 projects, the bulk of which may be as useful as Kleenex and worthy of expungement.

I would wholesale destroy it, except from my point of view there is stuff in there that might cost less to find than to recreate.

Even if the storage I am using on \<not my machines\> is ample and free, I feel bad about littering.

I have used the capability to download a single project zip. Can you offer a download zip of zips all projects?

I am guessing that the right person could automate this fairly rapidly and offer it for use. I am not that person.

urish, 2023-02-18:
Unfortunately, downloading all projects into one zip file not something trivial to implement. I've opened it for [voting](https://wokwi.com/features?sort=newest#feature-1590368776) - let's see how this goes

aaronjamt, 2023-02-20:
I feel like you could use Selenium to automate going through your entire project list and extract the Download ZIP link for each one, dump the links into a text file, then use something like Aria2c to bulk download all of them into a folder on your computer. I would try to build something like that but I don't want to hammer the servers unless one of the staff/admins/devs gives permission.

urish, 2023-02-20:
Cool idea, but the zip files are actually generated in the client side :) 

There's no server API for creating zipped projects. If this issue gets many votes and we start implementing it, we'll probably provide an API that allows you to request the creation of a single ZIP file with all your projects.

alto777, 2023-02-20:
Sorry. I understand how naive and annoying questions can be. I am optimistic and unschooled, so I thought that

something I could literally train a monkey to do using the browser menus

would be easy to automate. I watch wizards doing amazing things with one line of Unix, and so assumed that something like that would be not only possible but easy. For someone.

Find the next project. Open it. Use the download zip from the save menu. Rinse and repeat, 499 bottles of beer on the wall.

It would be easier if the three-dot menu on the My projects page had Download Zip in addition to Rename and Delete.

I don't know what Selenium and Aria2C are for or what they can do, but that sounds like it might be the kind of home-made solution I was thinking of.

As it happens, I have some monkeys here and I will train them and arrange a competition to see which one can press the right buttons in the right order fastest.

Not.

What I will do is whack away at it the hard way, 10 or 20 at a time when the juice for anything harder has spent until I have them all in captivity here. Then with any luck get around to deleting any that can be.

a7

aaronjamt, 2023-02-20:
I've made a PoC that is able to download all of my projects automatically, is it okay if I share it @urish? It downloads them all one-at-a-time rather than in parallel so it shouldn't hurt the servers too much (plus it has to load the homepage, click a project, then download, so any ratelimiting in place on the backend would affect it).

urish, 2023-02-20:
Yes, you can share it.

aaronjamt, 2023-02-20:
Ok, thanks. First download the correct version of the Chromedriver for the version of Chrome you have installed at the following link: https://chromedriver.chromium.org/downloads and extract the archive. You only need the `chromedriver` file (or `chromedriver.exe` on Windows). Next, download the Python script from https://pastebin.com/raw/75z1Lbrm. If you don't already have Python installed, install it from https://www.python.org/downloads. Open a terminal or command prompt and run `pip install selenium` then `cd path/to/the/script` and `python name-of-downloaded-file.py`. Follow the instructions the script provides to get your user ID and let it run.

Note: I'm assuming you have Google Chrome installed and are somewhat familiar with the command line. If you have Firefox, I'll see if I can tweak it to work with Firefox instead.

alto777, 2023-02-22:
_Good stuff_. This looks like a recipe I should be able to follow. I will try to come to a level of understanding before launching anything, and trust you will be able to help with that should I need it.

THX

a7

dela9116, 2023-04-20:
Aaronjmt:
I had to modify your code slightly to make it work on my computer with my chrome.  
It works great!  Much appreciated!!
My modified code:    [https://pastebin.com/raw/s08FzmE1](url)

drf5n, 2023-07-27:
> Ok, thanks. First download the correct version of the Chromedriver for the version of Chrome you have installed at the following link: https://chromedriver.chromium.org/downloads and extract the [...]

Short of using the script to do the downloading, you could open the page in Chrome and then use these xpath queries within the Developer Console to get the list of links and list of titles:

```
$x('//*[@id="__next"]/div/div/ul/li[*]/div[2]/div[1]/div[1]/a/@href').map(x => x.value)

$x('//*[@id="__next"]/div/div/ul/li[*]/div[2]/div[1]/div[1]/a').map(x => x.text)
```

With a bit of javascript, one could print a list of URLs and their titles and produce an easier-to manage version than a reverse-chronological list of images in an infinite-scrolling page.

You could save a local copy of your fully-scrolled [project page](https://wokwi.com/dashboard/projects) and do other processing of the data within using whatever language you like.

drf5n, 2024-02-01:
Here's a sloppy chunk of Chrome Console Javascript to make a HTML text listing of projects shown on your https://wokwi.com/projects page:

```
projectUrls=$x('//*[@id="__next"]/div/div/ul/li[*]/div[2]/div[1]/div[1]/a/@href').map(x=> x.value)
projectTitles=$x('//*[@id="__next"]/div/div/ul/li[*]/div[2]/div[1]/div[1]/a').map(x => x.text)

let myData ="";
for(ii = 0; ii< projectUrls.length; ++ii){
 let myAnchor = `<a href="https://wokwi.com${projectUrls[ii]}">${projectTitles[ii]}</a><br>\n`;
 myData =myData+myAnchor;
}
console.log(myData)
```

<a href="https://wokwi.com/projects/388535837079145473">UnoClockScope</a><br>
<a href="https://wokwi.com/projects/388498540754971649">DaveXWokwiListing</a><br>
<a href="https://wokwi.com/projects/388469991313392641">Fastled_777_blocking</a><br>
<a href="https://wokwi.com/projects/364610961628303361">forum230513_FastledRingStatemachine</a><br>

or Markdown:

```
projectUrls=$x('//*[@id="__next"]/div/div/ul/li[*]/div[2]/div[1]/div[1]/a/@href').map(x=> x.value)
projectTitles=$x('//*[@id="__next"]/div/div/ul/li[*]/div[2]/div[1]/div[1]/a').map(x => x.text)

let myMarkdown ="";
for(ii = 0; ii< projectUrls.length; ++ii){
 let myAnchor = `* [${projectTitles[ii]}](https://wokwi.com${projectUrls[ii]})\n`;
 myMarkdown =myMarkdown+myAnchor;
}
console.log(myMarkdown)
```

* [UnoClockScope](https://wokwi.com/projects/388535837079145473)
* [DaveXWokwiListing](https://wokwi.com/projects/388498540754971649)
* [Fastled_777_blocking](https://wokwi.com/projects/388469991313392641)

and a plain text CSV wouldn't be hard.

urish, 2024-06-30:
This issue hasn't received any votes in over a year, so closing it.

closed 2024-06-30 by urish