wokwi/wokwi-features #533: VSCode Extension Extremely Slow (RPI Pico)

aaronjamt, 2023-02-20:
**Describe the bug**
The VSCode extension (running an RPI Pico SDK program) runs at around 1% of realtime speed and the highest I've ever seen is 6% briefly and never again.

**To Reproduce**
This is the online version of the project I'm trying to run locally: https://wokwi.com/projects/357160621584861185

**Expected behavior**
I'd expect to see usable speeds (at least 25-50% or so, ideally 100%). Even in Firefox I've seen the Pico SDK at 100% speed, although with other Arduino based projects I've usually had terrible performance in Firefox compared to Chrome.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.10
 - No browser, using VSCode
 - VSCode is Version 1.75.1 with Electron 19.1.9 and Chromium 102.0.5005.194. The Wokwi extension is version 1.3.0 (the latest at the time of writing).

**Additional context**
I just installed the extension less than 30 minutes ago and have barely tried anything with Pi Pico in Wokwi before so it might very well be user error, I'm just not sure what I'm doing wrong. Also, it maxes out 1 CPU core and leaves the rest untouched, so that might be part of it. When I click the Pause or Stop button(s) in the simulation, it takes a few seconds to respond as well, so it's really struggling.

aaronjamt, 2023-02-20:
I realized that (in the browser) as soon as a PIO is enabled, the whole simulation slows down (Firefox drops from 100% to ~7-5% and Chrome drops to ~12-10%) so I added some delays to the beginning of my program but it doesn't seem to help. Even with a `sleep_ms(9999);` at the very beginning of my `int main()` method, as soon as the simulation starts in VSCode, the speed toggles between 1% and 0%. I timed it and 10 seconds after hitting start it had only simulated 0.133 seconds of real time (which is obviously not usable at all).

urish, 2023-02-20:
Thanks for the detailed report!

PIO can really affect the performance if it's running at full speed. Can you please try slowing down the PIO machine to let's say 1khz and see if this has any effect?

aaronjamt, 2023-02-20:
Do you mean using the `sm_config_set_clkdiv` function? The program isn't even getting to the point of starting PIO under VSCode. Here's the entire program I'm currently running that's super slow:
```
#include <stdio.h>
#include "pico/stdlib.h"
int main()
{
    sleep_ms(9999);
}
```

urish, 2023-02-20:
Which version of the Pico SDK are you using?

aaronjamt, 2023-02-20:
I'm using version 1.4.0 according to the numbers listed in `pico_sdk_version.cmake` in my SDK installation directory. Is that what you're referring to?

urish, 2023-02-20:
To make sure - can you run `git log` in your SDK installation directory and tell me what the first line says?

aaronjamt, 2023-02-20:
```
commit 2e6142b15b8a75c1227dd3edbe839193b2bf9041 (HEAD -> master, tag: 1.4.0, origin/master, origin/HEAD)
Author: graham sanderson <graham.sanderson@raspberrypi.com>
Date:   Wed Jun 29 22:56:27 2022 -0500

    SDK 1.4.0 release
```

Perfect timing, you commented right as I opened Github to check for a reply haha

urish, 2023-02-20:
So yeah, it's 1.4.0. Can you share the compiled ELF and UF2 files that you got?

aaronjamt, 2023-02-20:
Yes, here they are. Should I try to make a small example I can upload the source for, as well as the compiled files?
[snowflake.zip](https://github.com/wokwi/wokwi-features/files/10787509/snowflake.zip)

urish, 2023-02-20:
Yeah, that would be even better

aaronjamt, 2023-02-20:
Ok, will do. Might take a minute but I'll see what I can do.

aaronjamt, 2023-02-20:
Ok, I made a copy of the project and removed everything except the bare minimum to compile and run the project. I compiled it and ran the simulator to make sure it has the same issue (it does). I included all the build files exactly as they are from my machine, in case that helps. I'm running Linux and I saved the files to /tmp/snowflake so if you want to use the same path (and try a `diff -r` on the build directory or something) that's where to put everything. Hope this helps!
[snowflake.zip](https://github.com/wokwi/wokwi-features/files/10787549/snowflake.zip)

urish, 2023-02-20:
Thanks!

Can you try removing the following line from CMakeLists.txt and see if this helps?

```
pico_set_binary_type(snowflake no_flash)
```

aaronjamt, 2023-02-20:
Oh wait, I found the issue! I tried changing and removing options in `CMakeLists` and it turns out that `pico_set_binary_type(snowflake no_flash)` is the problem! I enabled that to minimize wear on the flash chip while working on a real Pi Pico but for some reason that causes the simulation to slow to a crawl

aaronjamt, 2023-02-20:
> Thanks!
> 
> Can you try removing the following line from CMakeLists.txt and see if this helps?
> 
> ```
> pico_set_binary_type(snowflake no_flash)
> ```

Yep, turns out that was it

aaronjamt, 2023-02-20:
Is that a known issue or should I make a new bug report specifically about `no_flash` causing the simulator to struggle?

urish, 2023-02-20:
As far as I can tell, Wokwi just ignores the data that is loaded to RAM. So no data is loaded into the simulator. We should probably investigate why this slows down the simulator so much.

How do you normally load `no_flash` binaries into the pico?

aaronjamt, 2023-02-20:
> As far as I can tell, Wokwi just ignores the data that is loaded to RAM. So no data is loaded into the simulator. We should probably investigate why this slows down the simulator so much.
> 
> How do you normally load `no_flash` binaries into the pico?

I believe you just upload the `.uf2` file normally. I haven't actually tried it because I haven't had access to a physical Pi Pico for a few weeks and, while reading some forum posts, found that line and added it to my code, made sure it compiled, and forgot about it. It's supposed to run from RAM until you powercycle the board, at which point it loses the firmware and you have to upload it again. I also tested the web version with `F1 -> Upload firmware` and it had the same issue with that line added.

urish, 2023-02-20:
Yes, `F1 -> Upload Firmware` was the first thing I tried too :-)

I've added an error message - if Wokwi can't find any flash data in the UF2 file, it'll now tell you to double check you haven't compiled with `no_flash`. Hopefully, this will help to prevent the same issue from occurring again in the future.

aaronjamt, 2023-02-20:
Awesome, thanks!

closed 2023-02-20 by urish