wokwi/wokwi-features #534: RPi Pico Error `WARN: starting new transfer on already active ep 0 out`

aaronjamt, 2023-02-21:
**Describe the bug**
Whenever I run a Pi Pico program (using the C/C++ SDK) that includes `stdio_init_all();`, it gives this error and frequently `printf` results in corrupted data being printed to the terminal.

**To Reproduce**
See the file I attached below. If you extract the archive and run `snowflake/snowflake.uf2` in Wokwi, it will show the error in the serial terminal. The source is included as well as the build files (assuming the archive is extracted to `/tmp/snowflake`). 

**Expected behavior**
I expect not to see this warning/error and, since this is a simulation not real hardware, there wouldn't be any data corruption issues (other than as a direct result of my code being buggy).

**Environment (please complete the following information):**
 - OS: Ubuntu 22.10
 - No browser, using VSCode
 - VSCode is Version 1.75.1 with Electron 19.1.9 and Chromium 102.0.5005.194. The Wokwi extension is version 1.3.0 (the latest at the time of writing). The RPi Pico SDK is version 1.4.0.

**Additional context**
[Here's a small program that reproduces this error, including the build files from my machine.](https://github.com/wokwi/wokwi-features/files/10798374/snowflake.zip)

aaronjamt, 2023-02-21:
When I run `stdio_uart_init();` _without_ `stdio_usb_init();` it doesn't have the `WARN: starting new transfer on already active ep 0 out` error but it still has data corruption issues. When I just use `stdio_usb_init();` the serial console never appears (so the simulated serial terminal must be UART, then, right?) However, even with that error not showing up, there's still data corruption in the serial monitor.

urish, 2023-02-22:
Thanks for reporting!

For seeing the `stdio` output, simply add the following connections to your diagram.json:

```
  "connections": [
    [ "pico:GP0", "$serialMonitor:RX", "", [] ], 
    [ "pico:GP1", "$serialMonitor:TX", "", [] ] 
  ],
```

(you can always copy them from https://wokwi.com/projects/new/pi-pico-sdk which uses this setup)

aaronjamt, 2023-02-22:
> Thanks for reporting!
>
> For seeing the `stdio` output, simply add the following connections to your diagram.json:
>
> ```
>   "connections": [
>     [ "pico:GP0", "$serialMonitor:RX", "", [] ],
>     [ [...]

I already have those lines and `printf` "works". It just has data corruption issues and that warning/error whenever it enables USB serial I/O.

aaronjamt, 2023-02-23:
> Thanks for reporting!
>
> For seeing the `stdio` output, simply add the following connections to your diagram.json:
>
> ```
>   "connections": [
>     [ "pico:GP0", "$serialMonitor:RX", "", [] ],
>     [ [...]

I already have those lines and `printf` "works". It just has data corruption issues with UART and that warning/error whenever USB is enabled