wokwi/wokwi-features #540: Controllino Boards *Feature*

Sandman1337, 2023-03-01:
Hi.
**Can you add Controllino boards?**
At my workplace, we use Controllino boards to test some of our products.
It is an industrialized version of Arduino's platform. For example, the **Controllino MEGA** uses the same chip as the **Arduino MEGA**, but are built very differently.
The Controllino also has relays built-in, and can be controlled with the library.
**If anything, the option to create and add the boards myself would be great!**

For information about Controllino: https://www.controllino.com/ 
[CONTROLLINO-MEGA-Pinout.pdf](https://github.com/wokwi/wokwi-features/files/10863405/CONTROLLINO-MEGA-Pinout.pdf)

I hope this would be possible, since this would make my life much easier at work.
Thanks!
[CONTROLLINO-MEGA-Pinout.pdf](https://github.com/wokwi/wokwi-features/files/10863405/CONTROLLINO-MEGA-Pinout.pdf)

Sincerly,
Simon

urish, 2023-03-01:
Hi Simon!

If you want to try adding the board yourself, you can try creating a custom board (SVG file + JSON file) with the pin descriptions. You can find an example + instructions for testing here:

https://github.com/wokwi/wokwi-boards#readme

The option to create custom boards is pretty new, so there's not much documentation, but you can learn by looking at the existing boards in that repo. Here to help!

Sandman1337, 2023-03-01:
Alright, thank you a lot for the quick reply!
I will look into this and try to create a custom Controllino board.
If I manage to create one that works well, I will share it here in this thread.
Wokwi looks very promising and will most differently help me with my projects!

Thank you for being so helpful!

closed 2025-01-13 by urish