wokwi/wokwi-features #541: Need more IC Modules in simulation panel.

ak4shp, 2023-03-03:
Please add some more ICs and Module in the simulation panel. 
A feature to import some kind of IC module externally will be a great addition.

ak4shp, 2023-03-03:
A few suggestions ->
1. RC522 RFID Reader
2. ULN2003 Stepper Motor Driver ( Yes there is already A4988 but this will be good add) 
3.

urish, 2023-03-04:
Thanks for the suggestions!

1. The [RC522 RFID](https://wokwi.com/features#feature-1242782423) is already open for voting (#359)
2. Feel free to open a separate issue for that specific driver (and we'll open it for voting).

Note that it's also possible to create your own ICs by writing a simulation model in C. See the Custom Chips API docs for examples: https://docs.wokwi.com/chips-api/getting-started

closed 2023-03-04 by urish