wokwi/wokwi-features #545: rfc2217ServerPort issue

realJoshByrnes, 2023-03-09:
Not sure where to put this as there doesn't seem to be a repo for the Wokwi VSCode Extension, but it relates the the rp2040js as that's the component I'm having issues with.

### wokwi.toml

```toml
[wokwi]
version = 1
firmware = './firmware/kaluma-rp2-pico-1.1.0-beta.1.uf2'
elf = ''
rfc2217ServerPort = 4000
```

### diagram.json

```json
{
  "version": 1,
  "author": "Josh Byrnes",
  "editor": "wokwi",
  "parts": [
    {
      "type": "wokwi-pi-pico",
      "id": "pico",
      "top": 0,
      "left": 0,
      "attrs": { "env": "kaluma-1.0.0" }
    }
  ]
}
```

## Issue

Current setup involves Wokwi Simular (VSCode) containing only a rPi Pico, and Kaluma (JS) firmware.

Whilst a connection can be established and REPL works as expected, it appears that some binary data is lost/encoded, causing a `kaluma flash` to never end. Everything else works as expected.

Notes:
* Kaluma works fine with real hardware via the COM port
* I have tried with multiple RFC2217 software such as [Perle's Trueport](https://www.perle.com/downloads/trueport.shtml) (recommended) and [Electronic Team's Serial to Ethernet Converter](https://www.eltima.com/products/serial-over-ethernet/) (commercial)

urish, 2023-03-11:
Update: couldn't get Trueport to work here. I'll try a different method and report.

realJoshByrnes, 2023-03-11:
I did post another software above, but it's only free for 14 days or similar.
https://www.eltima.com/products/serial-over-ethernet/

urish, 2024-05-02:
Closing, as I wasn't able to reproduce the issue.

closed 2024-05-02 by urish