wokwi/wokwi-features #549: ThingsBoard Issue

husseinprofile, 2023-03-18:
In file included from /libraries/ThingsBoard/src/ThingsBoard.h:25,
                 from sketch.ino:7:
/libraries/ThingsBoard/src/Telemetry.h:26:49: error: 'enable_if' in namespace 'ArduinoJson' does not name a template type
               typename = ARDUINOJSON_NAMESPACE::enable_if<ARDUINOJSON_NAMESPACE::is_integral<T>::value>>
                                                 ^~~~~~~~~
/libraries/ThingsBoard/src/Telemetry.h:26:58: error: expected '>' before '<' token
               typename = ARDUINOJSON_NAMESPACE::enable_if<ARDUINOJSON_NAMESPACE::is_integral<T>::value>>
                                                          ^
In file included from sketch.ino:7:
/libraries/ThingsBoard/src/ThingsBoard.h: In instantiation of 'bool ThingsBoardSized<MaxFieldsAmt, Logger>::sendKeyValue(const char*, T, bool) [with T = int; unsigned int MaxFieldsAmt = 8; Logger = ThingsBoardDefaultLogger]':
/libraries/ThingsBoard/src/ThingsBoard.h:654:37:   required from 'bool ThingsBoardSized<MaxFieldsAmt, Logger>::sendTelemetryInt(const char*, int) [with unsigned int MaxFieldsAmt = 8; Logger = ThingsBoardDefaultLogger]'
sketch.ino:122:56:   required from here
/libraries/ThingsBoard/src/ThingsBoard.h:1855:23: error: call of overloaded 'Telemetry(const char*&, int&)' is ambiguous
       const Telemetry t(key, value);
                       ^
In file included from /libraries/ThingsBoard/src/ThingsBoard.h:25,
                 from sketch.ino:7:
/libraries/ThingsBoard/src/Telemetry.h:43:12: note: candidate: 'Telemetry::Telemetry(const char*, float)'
     inline Telemetry(const char *key, float val)
            ^~~~~~~~~
/libraries/ThingsBoard/src/Telemetry.h:35:12: note: candidate: 'Telemetry::Telemetry(const char*, bool)'
     inline Telemetry(const char *key, bool val)
            ^~~~~~~~~

Error during build: exit status 1

husseinprofile, 2023-03-18:
> In file included from /libraries/ThingsBoard/src/ThingsBoard.h:25, from sketch.ino:7: /libraries/ThingsBoard/src/Telemetry.h:26:49: error: 'enable_if' in namespace 'ArduinoJson' does not name a template type typename = ARDUINOJSON_NAMESPACE::enable_if<ARDUINOJSON_NAMESPACE::is_integral::value>> ^~~~~~~~~ /libraries/ThingsBoard/src/Telemetry.h:26:58: error: expected '>' before '<' token typename = ARDUINOJSON_NAMESPACE::enable_if<ARDUINOJSON_NAMESPACE::is_integral::value>> ^ In file included from sketch.ino:7: /libraries/ThingsBoard/src/ThingsBoard.h: In instantiation of 'bool ThingsBoardSized<MaxFieldsAmt, Logger>::sendKeyValue(const char*, T, bool) [with T = int; unsigned int MaxFieldsAmt = 8; Logger = ThingsBoardDefaultLogger]': /libraries/ThingsBoard/src/ThingsBoard.h:654:37: required from 'bool ThingsBoardSized<MaxFieldsAmt, Logger>::sendTelemetryInt(const char*, int) [with unsigned int MaxFieldsAmt = 8; Logger = ThingsBoardDefaultLogger]' sketch.ino:122:56: required from here /libraries/ThingsBoard/src/ThingsBoard.h:1855:23: error: call of overloaded 'Telemetry(const char*&, int&)' is ambiguous const Telemetry t(key, value); ^ In file included from /libraries/ThingsBoard/src/ThingsBoard.h:25, from sketch.ino:7: /libraries/ThingsBoard/src/Telemetry.h:43:12: note: candidate: 'Telemetry::Telemetry(const char*, float)' inline Telemetry(const char _key, float val) ^~~~~~~~~ /libraries/ThingsBoard/src/Telemetry.h:35:12: note: candidate: 'Telemetry::Telemetry(const char_, bool)' inline Telemetry(const char *key, bool val) ^~~~~~~~~
> 
> Error during build: exit status 1

closed 2023-03-18 by husseinprofile

NgocThinhNTP18, 2023-03-28:
@husseinprofile 
Hi guy,
I have issue same than you. Did you find any solution yet ?

EliriaT, 2023-03-31:
Hi, I have the same issue. Did you solve the error somehow?

urish, 2023-03-31:
The error is due to a recent change in the ArduinoJson library: thingsboard/thingsboard-arduino-sdk#130

The work around is to use an older version of ArduinoJson. To do this, make sure you have the following lines in the "libraries.txt" file in your Wokwi project. Here's an example:

```
# Wokwi Library List
# See https://docs.wokwi.com/guides/libraries

ThingsBoard
ArduinoJson@6.20.1
```

Note that ArduinoJson@6.20.1 must come **after** ThingsBoard.

Complete project example: https://wokwi.com/projects/359639749716499457

EliriaT, 2023-03-31:
Thank you!  This solved the issue.

Manan-13, 2023-04-13:
> 

How do I solve this error in Arduino IDE ?