wokwi/wokwi-features #551: Neopixels with MicroPython on Pi Pico does not work as it should.

Lucvolders, 2023-03-19:
**Describe the bug**
On my fysical (real world) Pi Pico I am running a program that sets the pixels in a Neopixel ring.
If I run this program in a Wokwi simulation it does not work.
This is strange as it is a program from the official MicroPython documentantion.
Next to that there is another program for which I got a link on WokWi which is using
the same syntax and that runs flawlessly on a WokWi simulation with the ESP32

**To Reproduce**
This is the program that runs on an ESP32 but as the ESP32 is replaced by a Pi Pico it does not run anymore:
https://wokwi.com/projects/359662197132040193

This is a small program that runs on my fysiacal Pi Pico but not on a Wokwi simulation

```
import machine
from neopixel import NeoPixel

# Pin number connected to the neopixels
pin = machine.Pin(5, machine.Pin.OUT)

# Number of neopixels in the string
num_leds = 12
# Create a NeoPixel object with the pin and number of neopixels
np = NeoPixel(pin, num_leds, bpp=3)

# Set all neopixels to green
for i in range(num_leds):
    np[i] = (0, 0, 255)

# Update the neopixels to show the new brightness level and color
np.write()
```

**Expected behavior**
What should hapen is that the program when it starts sets all neopixels in the ring to green. But nothing happens.

In the simulation with the Pico the neopixels ring should display the rainbow colors but nothing happens.

**Environment (please complete the following information):**
 - OS: Windows 10
 - Browser [Firefox and Chrome tested
 

**Additional context**
Again strange as the program works on a real Pi Pico and follows the official MicroPython syntax.

urish, 2023-03-20:
Thanks for reporting! Initial investigation reveals that this is related to the SysTick timer implementation: the built-in neopixel library [uses the `bitstream` module](https://github.com/micropython/micropython-lib/blob/c8603192d1d142fdeb7f9578e000c9834669a082/micropython/drivers/led/neopixel/neopixel.py#L50), which, in turn, [uses systick for the timing-sensitive pulses](https://github.com/micropython/micropython/blob/711bac511e547096fe1e58af3ba70d31b244731f/ports/rp2/machine_bitstream.c#L60-L67).

Lucvolders, 2023-03-21:
I just saw that this was already listed as an issue on 28 december (issue 493)
whic describes the same problem.

closed 2023-03-22 by urish

urish, 2023-03-22:
Fixed! Thanks again for reporting.

Lucvolders, 2023-03-23:
Thanks !!!!
Just tested with some of my programs and it works flawless !!!

Luc