wokwi/wokwi-features #558: Reorganize "+ NEW PROJECT" to have Menu and Submenu

Vollbrecht, 2023-04-02:
**Is your feature request related to a problem? Please describe.**
when creating a new project on https://wokwi.com/dashboard/projects the list is getting bloated duo to the long list of rust and esp specific solutions. 
[image]

**Describe the solution you'd like**
i think it would be nice to have it into 2 steps -> "+ new project" -> category { arduino , esp-idf, micro-python, rust std. rust no_std ] -> than boards [ uno, esp.. etc] depending on previous selection

Vollbrecht, 2023-04-02:
or you can have it in reverse  -> First show the boards you support -> than the category / framework for that board that you support. Both have an appeal.

urish, 2023-04-16:
Thanks for the suggestion! What if we just added a search box on top, similar to what we have in the Add Part menu?

[image]

Vollbrecht, 2023-04-22:
I personally think a search bar is not a good solution here. For a new user - he needs to know in advanced what he has to search for - for a user that already crated a bunch of projects, it might fit it a bit better, but writing out something vs just have two clicks , i think the later option wins here every time. Though it just my personal gut feeling i am no professional ui designer.