wokwi/wokwi-features #559: Rust Esp32s3 board is missing

Vollbrecht, 2023-04-02:
**Describe the bug**
Creating a new project in https://wokwi.com/dashboard/projects misses the Esp32s3 option for Rust (std)

**To Reproduce**
Create new project here -> https://wokwi.com/dashboard/projects

**Expected behavior**
Because it is an option for Rust no_std and for arduino i would assume that Esp32s3 should also be there for Rust (std)

**Environment (please complete the following information):**
 - OS: Linux
 - Browser Firefox
 - Version 111

Vollbrecht, 2023-04-02:
From outside its hard to tell if this is a bug or a missing feature, so that's why i run with a bug.

urish, 2023-04-06:
Thanks for reporting! The thing is, we don't have an std builder for the ESP32-S3 yet. I'll discuss with @SergioGasquez to see what we can do about it.

closed 2023-04-12 by urish