wokwi/wokwi-features #560: Adafruit_GFX tft.fillCircleHelper(

Andr1961, 2023-04-05:
Hello, the example with tft.fillCircleHelper does not work, fillCircle draws.

#include "SPI.h"
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"

#define TFT_DC 9
#define TFT_CS 10
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

void setup() {
  tft.begin();
tft.fillCircle(50,100,50, ILI9341_GREEN);
tft.fillCircleHelper(50, 150, 50, 1, 20, ILI9341_RED);
}

void loop() { }

urish, 2023-04-05:
Hi, thanks for reporting! Have you tested the same code against real hardware? Does it work well there?

Andr1961, 2023-04-05:
yes, it works, fillCircleHelper does not want to use any combinations on the simulator.

urish, 2023-04-05:
Can you please send a photo of what you are seeing when you run that code in real hardware?

Also share a link to the Wokwi project where you tried this code?

Andr1961, 2023-04-05:
I'm sorry, I made a little mistake, I don't use "Adafruit_ILI9341.h", but ILI9341_Fast.h https://github.com/cbm80amiga/ILI9341_Fast it is not in the simulator, on http://www.instructables.com/id/Arduino-TFT-display-and-font-library / (Bodmer) it also works, but it is also not in the simulator, so it looks like the problem is in "Adafruit_ILI9341.h" , you can close it or I will close it myself, thanks, the simulator helps a lot, good luck)

closed 2023-04-05 by urish