wokwi/wokwi-features #564: tutorial to use in visual studio?

franciscogimeno2000, 2023-04-20:
Hello, does anyone know where I can find a tutorial from 0 on how to use Wokwi in visual studio Code? I am unable to get it to work and can't find anything. Thank you

urish, 2023-04-20:
@mattam can you please assist here?

franciscogimeno2000, 2023-04-20:
Hello, do you know where there is information from 0 to use the extension?

urish, 2023-04-20:
The documentation is at https://docs.wokwi.com/vscode/getting-started, and there's some video there - but it's not very comprehensive. Still, may be somewhat helpful.

franciscogimeno2000, 2023-04-20:
I've seen it and it doesn't explain anything to me. I'm sorry I'm clumsy. You only see the code loaded and not how it got to that point.

urish, 2023-04-20:
Yes the video there is not that great. Let's wait to see what @matititam can come up with.

franciscogimeno2000, 2023-04-20:
I use Visual Studio and I've never used Visual Studio Code... I don't know, it's a very strange application to handle.

franciscogimeno2000, 2023-04-20:
OK, thank you very much for your help. wait.

franciscogimeno2000, 2023-04-21:
> @franciscogimeno2000Había jugado con algunos proyectos en VSCode. Intentaré crear un video que muestre cómo hago que Wokwi se ejecute en VSCode. Espéralo para mañana :)

Hello, how is the tutorial going?

matititam, 2023-04-21:
It's cooking. Just a couple of hours. It will be ready

franciscogimeno2000, 2023-04-21:
ok

matititam, 2023-04-21:
Please have a look at this document. 
https://docs.google.com/document/d/1XQ6oEajXWh64AKem3sj6RQL7GPWH-MVa7xeWXK2pB6A/edit?usp=sharing

Share your feedback.

franciscogimeno2000, 2023-04-21:
OK thanks for the help.

Some questions.

How do you make it stop at a stopping point?

How are the values ​​of the variables monitored?

Could I load a *.ino file?

matititam, 2023-04-21:
To load an ino file, you will need Arduino Extension installed. 

https://marketplace.visualstudio.com/items?itemName=vsciot-vscode.vscode-arduino

Project: https://github.com/wokwi/arduino-lcd-helloworld

(download, Unzip, Load folder.. ) and follow the instructions in the ReadMe file. 

I had some hiccups setting it up. I will get back to you. Meanwhile, you, too, can give it a try. 

Regarding stopping at a particualr point -> It is not possible at this point. Same goes with monitoring variables (unless you prinr themout). I might be wrong ehre and I am positive that there are some solutions from @urish

franciscogimeno2000, 2023-04-21:
Opps... wow, I thought you could see and change the value of the variables, as well as put breakpoints in the program, without it being necessary to physically connect a micro.

closed 2025-01-13 by urish