wokwi/wokwi-features #565: ESP32 Simulation: FastLED.show() crashes when no of LEDs is a multiple of 8

github-ec, 2023-04-20:
**Describe the bug**
FastLED lib in combination with ESP32 here at Wokwi crashes 
if the number is exactly  8 or a multiple of 8. The first call of
FastLED.show() in loop() stops the whole sketch.
For one led-ring with 16 LEDs you can use 1 to 7 or 9 to 15 or 17 ...
but NUM_LEDS = 8, 16, 24, 32, 40 or 48 ... let the sketch crash ....

**To Reproduce**
Use the sketch
[https://wokwi.com/projects/362461890000450561](url)
but change the line 
`#define NUM_LEDS 49`    // This works!

to 

`#define NUM_LEDS 48` // This does not work, neither does 8, 16, 24, ... tested up to 48

**Expected behavior**
The number of LEDs should allow multiples of 8.

**Environment (please complete the following information):**
 - OS: Windows 11
 - Browser Google Chrome 
 - Version 112.0.5615.138 

**Additional context**
The identical sketch works with e.g. Arduino UNO simulation.
see [https://wokwi.com/projects/362458687561066497](url)

`#include <FastLED.h>
                        //FastLED lib in combination with ESP32 here at Wokwi crashes 
#define NUM_LEDS 49     // if the number is exactly  8 or a multiple of 8. The first call of
                        // FastLED.show() in loop() stops the whole sketch.
                        // For one led-ring with 16 LED you can use 1 to 7 or 9 to 15 or 17 ...
                        // but NUM_LEDS = 8, 16, 24, 32, 40 or 48 ... let the sketch crash ....
#define DATA_PIN 5

CRGB leds[NUM_LEDS];

void setup() { 
	Serial.begin(57600);
	Serial.println("resetting");
	FastLED.addLeds<WS2812B,DATA_PIN,GRB>(leds,NUM_LEDS);
  //FastLED.addLeds<NEOPIXEL, DATA_PIN>(leds, NUM_LEDS);  // GRB ordering is assumed
	FastLED.setBrightness(200);

}

void fadeall() { for(int i = 0; i < NUM_LEDS; i++) { leds[i].fadeToBlackBy(80); } }

void loop() {
  // put your main code here, to run repeatedly:

	for(int i = 0; i < NUM_LEDS-1; i++) {  // NUM_LEDS-1 as we use [i+1] in the loop()
    leds[i+1] = CRGB::Blue;            // Otherwise this index would become equal to NUM_LED
                                         // which is out-of-range of leds[NUM_LEDS] (0 ... NUM_LEDS-1)
    leds[i] = CRGB::Yellow;
    Serial.println(i);
    FastLED.show(); 
    fadeall();  
    delay(200);
  }

}`

urish, 2023-04-23:
Thanks for reporting! Weird indeed

urish, 2023-04-23:
Interestingly, this doesn't happen with ESP32-C3: https://wokwi.com/projects/362801722531028993

urish, 2023-04-29:
Fixed!

closed 2023-04-29 by urish