wokwi/wokwi-features #566: Create a static svg logo endpoint, to serve Badges, for inclution into GH etc

Vollbrecht, 2023-04-21:
**Is your feature request related to a problem? Please describe.**
Allow users to include Badges on there GH pages, without the need for them to include a copy of the wokwi logo.

**Describe the solution you'd like**
They are dozen approaches to it. One simple way that also uses a caching service is, providing an static json response somewhere on wokwi.com/logo or on a different subdomain something like logo.wokwi.com. This json response could look something like 
```json
{
  "schemaVersion": 1,
  "label": "left site of the badge txt", -- keep this empty when using an actual logo
  "message": "right site txt of the logo", -- something like "Simulate this in Wokwi"
  "color": "orange", -- color of the message text
  "logoSvg":"An SVG string containing your logo",
}
```
this than can be used with the endpoint service of shield.io 
A complete list of the json format + all the advantages are listed here ->  https://shields.io/endpoint

urish, 2023-04-21:
So something like this?

```json
{
  "schemaVersion": 1,
  "label": "",
  "message": "Simulate in Wokwi",
  "color": "#3166bb",
  "logoSvg": "<svg xmlns=\"http://www.w3.org/2000/svg\" viewBox=\"0 0 19.816 16.406\" height=\"62.007\" width=\"74.894\">\n <path d=\"M4.192 15.524c-.63-.738-.765-1.863-.32-2.684.405-.748.346-1.136-.583-3.84-.937-2.73-1.116-3.05-1.935-3.455-1.916-.95-1.765-3.453.242-4.001C3.488 1.027 4.97 2.93 4.014 4.649c-.416.75-.36 1.113.593 3.866.97 2.796 1.126 3.071 1.964 3.456 1.192.549 1.62 1.884.969 3.025-.65 1.14-2.566 1.441-3.348.528zm1.912-1.003c.465-.38.426-.796-.103-1.098-.238-.136-.544-.186-.681-.112-.423.23-.407 1.014.025 1.26.226.13.567.107.76-.05zM2.612 4.164c.346-.189.224-1.078-.18-1.308-.337-.193-.995.268-.98.686.013.387.809.813 1.16.622zM11.945 14.994c-.63-.738-.765-1.863-.32-2.684.405-.748.346-1.136-.583-3.84-.937-2.73-1.116-3.05-1.935-3.455-1.916-.95-1.765-3.453.242-4.001C11.24.497 12.723 2.4 11.767 4.119c-.416.75-.36 1.113.593 3.866.97 2.796 1.126 3.071 1.964 3.456 1.192.549 1.62 1.884.969 3.025-.65 1.14-2.566 1.441-3.348.528zm1.912-1.003c.465-.38.426-.796-.103-1.098-.238-.136-.544-.186-.681-.112-.423.23-.407 1.014.025 1.26.226.13.567.107.76-.05zM10.365 3.634c.346-.189.224-1.078-.18-1.308-.337-.193-.995.268-.98.687.013.386.809.812 1.16.62z\"/><path d=\"M4.543 16.214c-2.027-1.187-1.3-4.185 1.015-4.185.75 0 .816-.177 1.864-1.169 1.047-.992 2.833-2.627 4.035-3.222.324.46.571 1.615.607 1.748-.772 0-2.75 1.515-3.588 2.538-.81.988-.985 1.4-.928 2.188.052.73-.081 1.11-.541 1.54-.78.728-1.787.958-2.464.562zm1.56-1.695c.215-.56-.025-.903-.634-.903-.594 0-.896.41-.693.94.198.515 1.125.49 1.327-.037zM13.185 12.054s.284-.565 1.392-3.717c1.44-4.078 1.509-4.407 1.096-5.263-.887-1.84.835-3.684 2.691-2.88.694.3.931.585 1.236 1.49.332.983.303 1.19-.244 1.72-.341.33-.896.716-1.231.856-.47.196-.972 1.255-2.186 4.618-1.443 3.996-1.06 2.817-.67 3.541.961 1.79-.26.492-2.083-.365zm5.066-9.85c-.006-.6-.355-.83-.921-.605-.255.1-.486.309-.513.462-.085.473.537.951 1 .768.241-.096.437-.377.434-.625z\"/>\n</svg>"
}
```

Vollbrecht, 2023-04-21:
yeah i think looks good. Here are two links to a real example : 
https://raw.githubusercontent.com/chris48s/shields-endpoint-test/2958b1745abdc566152d54d336550a29c1ccb553/badge1.json
https://img.shields.io/endpoint.svg?url=https%3A//raw.githubusercontent.com/chris48s/shields-endpoint-test/2958b1745abdc566152d54d336550a29c1ccb553/badge1.json

urish, 2023-04-22:
https://wokwi.com/badge/simulate-in-wokwi.json → [image]

Though, I'm not very happy with the contrast of the logo against the background

Vollbrecht, 2023-04-22:
yeah, it would be nice if we can set the logo to white, on a black/greay background - default look for all the logos + badges is something like -> 
[image]
or without dark-mode how it is on esp-hal -> 
[image]

Vollbrecht, 2023-04-22:
ah i just found out, if the label is empty string the color is used for both right and left side. it looks like this if for example i set it to Wokwi.
[image]

urish, 2023-04-22:
Getting the logo in white is as simple as adding a fill to the svg string. Based on the examples in your previous comment, the right side displays some status or information (build status, version number, count of users, etc)., but I can't come up with a good idea of what to display there (other than something like "Run simulation" or "simulate this project")

Vollbrecht, 2023-04-22:
naming is hard :smile: some more ideas assuming left text is wokwi:   "Run" , "Click to Simulate",  "Open", "get started", "play with", "run in browser"

Vollbrecht, 2023-04-22:
here is the svg string with withe filling, though is this the correct or maybe some more grayish white? 
```
<svg xmlns="http://www.w3.org/2000/svg" viewBox="0 0 19.816 16.406" height="62.007" width="74.894"> <path fill="#FFFFFF" d="M4.192 15.524c-.63-.738-.765-1.863-.32-2.684.405-.748.346-1.136-.583-3.84-.937-2.73-1.116-3.05-1.935-3.455-1.916-.95-1.765-3.453.242-4.001C3.488 1.027 4.97 2.93 4.014 4.649c-.416.75-.36 1.113.593 3.866.97 2.796 1.126 3.071 1.964 3.456 1.192.549 1.62 1.884.969 3.025-.65 1.14-2.566 1.441-3.348.528zm1.912-1.003c.465-.38.426-.796-.103-1.098-.238-.136-.544-.186-.681-.112-.423.23-.407 1.014.025 1.26.226.13.567.107.76-.05zM2.612 4.164c.346-.189.224-1.078-.18-1.308-.337-.193-.995.268-.98.686.013.387.809.813 1.16.622zM11.945 14.994c-.63-.738-.765-1.863-.32-2.684.405-.748.346-1.136-.583-3.84-.937-2.73-1.116-3.05-1.935-3.455-1.916-.95-1.765-3.453.242-4.001C11.24.497 12.723 2.4 11.767 4.119c-.416.75-.36 1.113.593 3.866.97 2.796 1.126 3.071 1.964 3.456 1.192.549 1.62 1.884.969 3.025-.65 1.14-2.566 1.441-3.348.528zm1.912-1.003c.465-.38.426-.796-.103-1.098-.238-.136-.544-.186-.681-.112-.423.23-.407 1.014.025 1.26.226.13.567.107.76-.05zM10.365 3.634c.346-.189.224-1.078-.18-1.308-.337-.193-.995.268-.98.687.013.386.809.812 1.16.62z"/>
<path fill="#FFFFFF" d="M4.543 16.214c-2.027-1.187-1.3-4.185 1.015-4.185.75 0 .816-.177 1.864-1.169 1.047-.992 2.833-2.627 4.035-3.222.324.46.571 1.615.607 1.748-.772 0-2.75 1.515-3.588 2.538-.81.988-.985 1.4-.928 2.188.052.73-.081 1.11-.541 1.54-.78.728-1.787.958-2.464.562zm1.56-1.695c.215-.56-.025-.903-.634-.903-.594 0-.896.41-.693.94.198.515 1.125.49 1.327-.037zM13.185 12.054s.284-.565 1.392-3.717c1.44-4.078 1.509-4.407 1.096-5.263-.887-1.84.835-3.684 2.691-2.88.694.3.931.585 1.236 1.49.332.983.303 1.19-.244 1.72-.341.33-.896.716-1.231.856-.47.196-.972 1.255-2.186 4.618-1.443 3.996-1.06 2.817-.67 3.541.961 1.79-.26.492-2.083-.365zm5.066-9.85c-.006-.6-.355-.83-.921-.605-.255.1-.486.309-.513.462-.085.473.537.951 1 .768.241-.096.437-.377.434-.625z"/></svg>
```

urish, 2023-04-22:
https://wokwi.com/badge/click-to-simulate.json -> 

[image]

Vollbrecht, 2023-04-22:
> https://wokwi.com/badge/click-to-simulate.json ->
> 
> [image]

looks good to me, i added it into the updated readme draft

Vollbrecht, 2023-04-22:
ah dang you renamed the link, just used the other one 🤦

Vollbrecht, 2023-04-22:
some small other thing i noted.
wokwi symbol does not look good in github darkmode.
[image]
[image]
I tried looking into organisation settings but could not find a place where you can upload one for light and dark mode. is there such a setting. Though its just a minor nit here.

closed 2023-04-27 by urish