wokwi/wokwi-features #567: ESP32C3 Incorrect Timer Duration in Wokwi

apollolabsdev, 2023-04-26:
Rust timeout duration for timer instances for the ESP32C3 appears to double in the Wokwi simulation. 

The code example under the link below instantiates a timer to toggle an LED every 2 seconds. In the simulation, the LED is toggled every 4 seconds instead.

Link to Project Example: https://wokwi.com/projects/363068349808311297

The same code was verified on ESP32C3 hardware by @Sergio Gasquez and produced the expected toggle duration of 2 seconds.

urish, 2023-04-26:
Issue reproduced successfully

urish, 2023-04-27:
Update: looked into this. Seems like we get the wrong CPU (and APB) clock frequency. Rust code says it should be 80MHz:
https://github.com/esp-rs/esp-hal/blob/a950a3ba3215bba9c8b3a32e017601f0eb6574e9/esp-hal-common/src/clock/mod.rs#L336

However, TRM says it should be 20MHz (if I understood correctly): the `SYSTEM_SOC_CLK_SEL` field of `SYSTEM_SYSCLK_CONF_REG` equals 0 (the reset value), which means the CPU clock is derived from the XTAL clock:

[image]

Then, `SYSTEM_PRE_DIV_CNT` equals 1, which means the frequency should be 20 MHz:

```
CPU_CLK = XTAL_CLK/(SYSTEM_PRE_DIV_CNT + 1)
SYSTEM_PRE_DIV_CNT ranges from 0 ~ 1023. Default is 1
```

So `40/(1+1)=20`. 

Not sure whether this is me misunderstanding something or TRM is wrong about how the CPU clock is derived when `SYSTEM_SOC_CLK_SEL` = 0.

Note: after fixing the implementation to match the TRM, we now get the LED to toggle every 8 seconds. Which is in par with the discrepancy in the CPU clock value (20MHz on Wokwi vs 80MHz in real chip, assuming at least the Rust code is correct).

apollolabsdev, 2023-04-27:
Revising the TRM, I have the same understanding that the CPU clock defaults to 20MHz at boot. If that is the case, this makes the clock defaults in the Rust code incorrect. I would think it makes more sense to adjust it there, no?

However, on another note, per the provided code the LED is supposed to toggle every 2 seconds not 8 (which was 4 in the issue). This doesn't seem to coincide with speeding up the clock source.

bjoernQ, 2023-04-27:
Just did a quick-check with both esp-idf 2nd stage-bootloader and direct-boot mode on real hardware: in both cases the LED toggles every 2 seconds for me.

Might be the TRM talks about what the HW initializes with but the ROM code change it - needs more investigation

urish, 2023-04-27:
> Might be the TRM talks about what the HW initializes with but the ROM code change it - needs more investigation

Thanks for looking into this @bjoernQ! 

I did a quick test, connecting to the physical chip with GDB, restarting the chip, and printing the value of SYSTEM_SOC_CLK_SEL before ROM executes, and it was 0. 

```
0x40000000 in ?? ()
(gdb) x/x 0x600C0058
0x600c0058:     0x000a8001
(gdb) p/x (*0x600C0058 >> 10)&0x3
$1 = 0x0
(gdb)
```

(0x600C0058 is `SYSTEM_SYSCLK_CONF_REG`, and bits 10..11 are `SYSTEM_SOC_CLK_SEL`)

Can you try to print the value and see if it differs during the 2nd stage bootloader or in direct boot mode?

bjoernQ, 2023-04-27:
in main I see
`00000000000010101000000000000001`
with direct-boot

with 2nd stage bootloader I see
`00000000000010101000010000000000`

So, the second-stage bootloader definitely changes things

But interesting the timing is right (on hardware) in both cases

urish, 2023-04-27:
Thanks! 

So if I understand correctly, you see that regardless of whether `SYSTEM_SOC_CLK_SEL` is 0 or 1, the clock still behaves the same (as if CPU CLK is 80MHz in both cases)?

urish, 2023-04-27:
Update: until we have some better information, I changed the behavior of `SYSTEM_SOC_CLK_SEL` to match what we observed - setting it to either 0 or 1 will use the PLL as a clock source for the CPU. This should solve the issue here.

I also pinged the Espressif documentation team regarding this and waiting for their response.

bjoernQ, 2023-04-28:
Thanks! Would be great if you can update this issue if there is more information available (and we do the same if we gather further information regarding this)

songruo, 2023-05-04:
@urish I saw your email to the Espressif documentation team. I can probably help to clarify some of your confusion.

The TRM is correct. And yes, the default CPU clock frequency at chip reset is 20MHz (XTAL 40MHz / (1 + 1)). In ESP-IDF, at 2nd stage bootloader, the CPU clock source is switched to PLL, so CPU clock frequency is raised to 80MHz. And then, by default menuconfig option, at application startup stage, the CPU clock frequency is further raised to 160MHz (clock source is still PLL).

Let me know if you have any other question that I can help to answer for the debugging!

songruo, 2023-05-05:
The logic in ESP-IDF is as following:

At power on reset, the `RTC_APB_FREQ_REG` is 0. Then at 2nd stage bootloader, we will check whether value in `RTC_APB_FREQ_REG` is equal to 80MHz. If not, then we will do CPU clock source switch to let CPU frequency be 80MHz, and store the current APB clock frequency, which is 80MHz, into `RTC_APB_FREQ_REG`, since then, `RTC_APB_FREQ_REG` will start to reflect the correct APB clock frequency; If value in `RTC_APB_FREQ_REG` already equals to 80MHz at the initial check (this could happen at CPU reset etc. cases), then CPU clock source switch operation is skipped.

urish, 2023-05-05:
Thanks for explaining! Uploaded a fix that matches what you described. So now `SYSTEM_SOC_CLK_SEL` is implemented in the way described in the `TRM`, and `RTC_APB_FREQ_REG` is initialized to 0 and you described (and then the bootloader takes care of the rest).

The original project link is broken, so I reuploaded the code to https://wokwi.com/projects/363880815393002497 and tested it, and now the LED toggles every 2 seconds as it should. So we're good :)

closed 2023-05-05 by urish