wokwi/wokwi-features #569: Simulating Access Point Scanning Crashes the simulation

Vollbrecht, 2023-04-27:
**Describe the bug**
using the AccesPoint scanning method in rust std esp-idf-svc -> scan() behaves differently in simulation as on real hardware.

**To Reproduce**
Here is a minimal repository that can build a minimal example : https://github.com/Vollbrecht/wifi-scan-wokwi-bug

**Expected behavior**
Dont return an ESP-FAIL error -> Either find the only available acces-point "Wokwi-Guest" or doesnt find any but continoue with the simulation

**Additional context**
Log output from simulation and real hardware:

[hardware_output.txt](https://github.com/wokwi/wokwi-features/files/11347708/hardware_output.txt)
[simulation_output.txt](https://github.com/wokwi/wokwi-features/files/11347709/simulation_output.txt)

urish, 2023-04-27:
Thanks! Can you please add the compiled .elf file to the repo? I'm not very good at compiling Rust, so I'm bound to spend an hour or so just to figure out how to compile it.

urish, 2023-04-27:
(You can also attach the compiled ELF file to this issue, if that's easier)

Vollbrecht, 2023-04-27:
> (You can also attach the compiled ELF file to this issue, if that's easier)

done

urish, 2023-04-27:
Thanks! Issue reproduces for me

urish, 2023-04-28:
Initial investigation shows that the simulated program is sending an ARP packet, without even connecting to the access point:

[image]

urish, 2023-04-28:
Trying to run the same on physical ESP32-C3, it also fails at the same point:

[image]

My current bet is that there's some issue with the NVS partition. Can you try to dump the current partition table and the NVS partition from your chip?

Vollbrecht, 2023-04-28:
ah i forget about the nvs partition - i flashed not on a brand new one esp and didn't override it :D will check it out

urish, 2023-05-04:
Any updates? Should we keep this open?

Vollbrecht, 2023-05-05:
Yeah i couldn't pinpoint it down exactly, to a repeatable case. Was using different methods to work with the nvs. I was also playing with a newer version of esp-idf here. If it surface again and i can pinpoint it better i will reopen.

closed 2023-05-05 by Vollbrecht