wokwi/wokwi-features #570: The ESP32S3 Servo does not work properly.

Zhentao-Lin, 2023-04-30:
The ESP32S3 Servo does not work properly.
This is my project: https://wokwi.com/projects/363419673189621761

urish, 2023-04-30:
Thanks for reporting! 

This is due to the missing LEDC PWM peripheral in the ESP32-S3: https://docs.wokwi.com/guides/esp32#simulation-features

Zhentao-Lin, 2023-04-30:
https://wokwi.com/projects/363243142349897729
https://wokwi.com/projects/363431350256084993
Here are my two examples, which I have tested on ESP32S3. But in the case of Servo, the simulation software is still not good.

urish, 2023-04-30:
I think I didn't explain it correctly - the LEDC PWM peripheral is *not yet* simulated in Wokwi for the ESP32-S3. It's totally working and functional on the actual hardware. It's a missing feature in the simulation.

Zhentao-Lin, 2023-04-30:
Ok, so how do I use my S3 driver Servo on an emulator？
Thank you for your quick reply.

Zhentao-Lin, 2023-04-30:
I'm looking forward to seeing this feature added.

urish, 2023-04-30:
For now, the only way would be to use it with ESP32 (or S2/C3), until we get to implement LEDC PWM for the ESP32-S3.