wokwi/wokwi-features #574: Weird display on Oled SSD1306 on vscode-extension

AnisBoubaker, 2023-05-03:
**Describe the bug**
When displaying information on an SSD1306 Oled, the result is weird (as if the display was divided into 4 different zones).  The problem does not occur on the website, only on the VS extension. 

**To Reproduce**
The simulated project is this one: https://wokwi.com/projects/363172552908797953
The same code is simulated on vscode. The elf and bin files are attached. 

**Expected behavior**
I expect the display to work the same as on the website. 

**Environment (please complete the following information):**
 - OS: MacOS
 - Browser: Not relevant (vscode)
 - VSCode Version: 1.77.3

urish, 2023-05-03:
Missing bin/elf files

AnisBoubaker, 2023-05-03:
> Missing bin/elf files

You were too fast :-) Just attached them to the initial comment.

urish, 2023-05-03:
Thanks! Seems like the issue is in the firmware - loading the compiled firmware to the website (go to https://wokwi.com/projects/363172552908797953, press F1 in the code editor, select "Upload Firmware and Start Simulation...", and choose firmware.bin) gives the same result as in VS Code. 

How do you compile the firmware?

AnisBoubaker, 2023-05-03:
With PlatformIO, Arduino framework.  Here is the content of the platform.ini content: 

[env:esp32]
platform = espressif32
framework = arduino
board = esp32dev
lib_deps = 
	adafruit/Adafruit SSD1306@^2.5.7
	madhephaestus/ESP32Servo@^0.13.0

urish, 2023-05-03:
Can you please share a repo with the complete setup?

AnisBoubaker, 2023-05-03:
Here you are: https://github.com/AnisBoubaker/WokwiTest

urish, 2023-05-04:
Thanks! I clone your repo, ran `pio run` to build the code, and then opened the folder in VS Code. Seems that it works correctly:

[image]

Note: I had espressif32 version 5.2.0 installed. I'm now upgrading to the latest version (6.2.0), to see if I get a different result.

urish, 2023-05-04:
Getting the same results even after upgrading the espressif32 platform.

Can you please run `pio pkg update`, then delete the `.pio` build directory, and run `pio run` to rebuild the project, and see if this changes anything for you?

AnisBoubaker, 2023-05-04:
That fixed it, thank you for your help! I had an old version of expressif32, sorry I thought all these packages were always the latest version unless I say so.

closed 2023-05-04 by AnisBoubaker