wokwi/wokwi-features #578: Button release doesn't work properly

kontiko, 2023-05-16:
**Describe the bug**
Button doesn't release when Mouse is moved from Button

**To Reproduce**
Press on Button and hold it.
Move Mouse away from Button.
Release Mouse Button.
Button is still active

**Expected behavior**
Button releases even when Mouse moved

**Environment (please complete the following information):**
 - OS: Linux
 - Browser Firefox
 - Version 113.0

urish, 2023-05-21:
This started as a bug, but AFAIK some users are now using it as an "undocumented feature" to hold multiple buttons at a time. Not sure if we should fix it - I'll post this issue to the [Wokwi discord chat](https://wokwi.com/discord) to get more opinions.

urish, 2023-05-21:
Note that there's also a documented keyboard shortcut to achieve the same functionality: https://docs.wokwi.com/parts/wokwi-pushbutton#stickiness

Vollbrecht, 2023-05-21:
I personally think than it should be fixed. And for the people who needed that functionality. They should either using a dip switch - or you could add an option to the button itself  - to make it a latching button.  For a speed-runner its not nice if you fix a glitch but for a normal player the overall experience is more important. And i think serving the normal player here is the right call.
I think one root of the problem is that the people didn't discover the latching option. So i think it would be nice if you add it directly as a toggle option here:
 
[image]

And maybe calling it a latching button and not a sticky one :smile_cat: Though i admit it is a bit funny.

AriellaE, 2023-06-29:
Hello @kontiko 
I fixed this bug. 
Thank you very much!

closed 2023-06-29 by AriellaE