wokwi/wokwi-features #581: Unable to find files on SD card when using Visual Studio Code Plugin

Marcus10110, 2023-05-23:
I'm unable to access any files from the virtual SD Card when running the simulator from VS code.

Specifically, the ls command prints out an empty list:

```
Files on card:
   Size    Name
```

However, if I restart the simulator using the restart button, I see diagram.json.

```
   Size    Name
      1496 diagram.json
```

Refreshing after that continues to just show diagram.json.

I have quite a few other files in the *.ino directory.

I've been testing with this project, both in the webapp and in vs code locally. This works great in the webapp.
https://wokwi.com/projects/365487016657699841

urish, 2023-05-23:
I can confirm the VS Code plugin does not support loading files into the SD Card at the moment.

Marcus10110, 2023-05-23:
Thanks @urish!

I'd love to see this on the feature voting site then.

Marcus10110, 2023-05-24:
For background, I'm very excited to use wokwi to streamline the development process for embedded UIs. There are a handful of other options for rapid prototyping embedded UI, such as implementing the high level UI code in a cross platform way and targeting x86 for development, but the ability to just simulate the hardware should hopefully be much less overhead. I'm hoping to get to the point where I can recompile and simulate in seconds, on every save - just like web development.

A key piece of of this workflow is to quickly and easily add images - ideally by just pasting new images into a data folder and then immidiately accessing them from code.

The next step after this would be to automatically grab screenshots from the simulated LCD screen, and automate the process of displaying and screen-grabbing every screen. Then I could easily compare all the screens with figma prototypes, somewhat like how storybook.js can display react components in an easy to develop way.

I'm really enjoying wokwi so far! Thanks to the whole team for creating such a great tool.

urish, 2023-05-26:
Interesting, thanks for explaining! Just had a discussion with the engineers of [Embedded Wizard](https://www.embedded-wizard.de/) this week, are you familiar with their product?

> The next step after this would be to automatically grab screenshots from the simulated LCD screen, and automate the process of displaying and screen-grabbing every screen. Then I could easily [...]

It's already possible with the Wokwi CLI (through GitHub actions). It's not yet documented, but I can create a quick example and share it on GitHub.

Based on your questions / suggestions, it seems like you have an interesting use case. If you'd like to hop on a call and discuss it, I'd love to.

Phetdanai, 2023-12-22:
But, I'm not Wokwi Clubs member.
How can I insert file(s)?

Phetdanai, 2023-12-22:
Example code for ESP32 :

[SD_Test.txt](https://github.com/wokwi/wokwi-features/files/13748567/SD_Test.txt)

maks, 2024-09-25:
@urish I'm excited by the possibility of using the VSCode extension to speed up development work on our [picoTracker](https://github.com/xiphonics/picoTracker) project. Having a sdcard is essential as currently our firmware has a hard requirement on a sdcard for storing at a minimum its config and project files, though I could put some hacks in to workaround that, reading/writing to the sdcard is actually some of the dev work I'd like to use wokwi to speed up the dev cycle with! 🙂

wallysalami, 2026-09-14:
Is there any news about this feature?

urish, 2026-09-29:
Update: some unexpected things have happened in the past 6 months, which have greatly shifted our roadmap. This feature is still high on the priority list, and we expect to start working on it in a week, and to deliver it during Q4.

urish, 2026-10-01:
Good news: SD card support for Wokwi for VS Code is out in version 4.0.0 🎉

You can set it up in two ways:

- Simple: Put your files in an `sdcard/` folder next to `wokwi.toml` and they are on the card when the simulation starts. No configuration needed.
- Advanced:  Add an `[[sdcard]]` section to `wokwi.toml`: pick a different `folder` or a raw disk `image` (e.g. dumped from a real card), set the card `size` (default 8M, up to 256M), and turn on `writeback` to have whatever the firmware writes copied back to your folder/image when the simulation stops or restarts (#605).

Docs + Wokwi CLI support (same `wokwi.toml` config, plus `--sdcard` flag for CI) are coming next, in a few days.

urish, 2026-10-02:
Documentation and CLI supports are also ready:

- https://docs.wokwi.com/vscode/project-config#sd-card-contents
- https://docs.wokwi.com/wokwi-ci/cli-usage#sd-card

Enjoy!

closed 2026-10-02 by urish