wokwi/wokwi-features #584: Newer gcc compiler

github-rkaiser, 2023-05-26:
Are there plans to provide wokwi with a newer gcc compiler? Currently we have gcc version 8.4.0 for an esp32 project.

Thanks
Richard

urish, 2023-05-26:
We do not set up the gcc compiler - rather, it comes from the [https://github.com/espressif/arduino-esp32](ESP32 Arduino Core), which, in turn, uses the esp-idf toolchain.

AFAIK they are currently migrating from esp-idf 4.4 to esp-idf 5.1, so I assume a newer gcc version will be available once they finish the migration and release a new version of the ESP32 Arduino core.

I hope this helps!

urish, 2023-05-26:
Note that you can also build the projects on your own computer, with newer esp-idf version (and hence newer GCC), and then use Wokwi for VS Code just for the simulation: https://docs.wokwi.com/vscode/getting-started

closed 2023-05-26 by urish

github-rkaiser, 2023-05-26:
After checking the espressif documentation, I still have one more question. In

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/gcc.html#gcc-version

it is stated

> The previous GCC version was GCC 8.4.0. This has now been upgraded to GCC 11.2.0 on all targets. Users that need to port their code from GCC 8.4.0 to 11.2.0 should refer to the series of official GCC porting guides listed below

So why do I still get version 8.4.0 by

```
Serial.print("GCC Version: ");
Serial.print(__GNUC__); Serial.print(".");
Serial.print(__GNUC_MINOR__);  Serial.print(".");
Serial.print(__GNUC_PATCHLEVEL__);
```
Do I have to change any settings? Or do I just have to wait some time until this upgrade takes place?

Thanks 
`Richard`

urish, 2023-05-26:
Hi Richard,

If you are compiling on Wokwi, then this uses the Arduino ESP32 Core, which has not *yet* updated to ESP-IDF 5.x. The docs are are referring to belong to ESP-IDF 5.0. So you either have to wait until the Arduino ESP32 Core releases a new version, or use the esp-idf directly (and compile the code on your computer).