wokwi/wokwi-features #585: ESP32: PWM output incorrect for high duty values

jonschz, 2023-05-26:
**Describe the bug**
The simulated ESP32 outputs incorrect PWM values at or slightly below 100 % output.

**To Reproduce**
Simulation project: https://wokwi.com/projects/365803776256601089

1. Run the project for 3-4 seconds.
2. Analyse the output of the Logic Analyzer.

**Expected behavior**
`analogWrite(18, 4095)` and `analogWrite(18, 4096)` should output near full or full power, respectively.

**Actual behavior**
Above a duty value of 4094, the PWM output drops to about 50%:
[image]

**Environment:**
 - OS: Windows 10
 - Browser: Firefox / vscode addon
 - Version: 133 / v1.3.3

**Additional context**
This issue also appears in firmware binaries using `esp-idf`. Notably, there is an error in the [documentation](https://github.com/espressif/esp-idf/pull/11516) for the ESP32, which may or may not be related in the present issue.

In firmware binaries, this issue also appears when the resolution is set to 10 bits and the duty value is set above 1023 (where I again find roughly 50% output instead of 100%).

Related discussion: https://github.com/arendst/Tasmota/issues/18726#issuecomment-1564185309

urish, 2023-05-26:
Thanks for the detailed report!

Have you had a chance to test it on actual hardware?

jonschz, 2023-05-26:
Not personally, but the firmware binaries I used have been tested on actual hardware by other people. See the thread I linked: https://github.com/arendst/Tasmota/issues/18726#issuecomment-1564185309. Further note the comment regarding the maximum value of 1024 instead of 1023 (which would be 4096 in the present case).

jonschz, 2023-05-28:
I should add that I don't own an ESP32, but only an ESP8266 which is not supported on Wokwi right now. Tasmota, which is the software I am working on, supports both chips, but I can't test any ESP32 binaries on real hardware at the moment. It should still be quite clear that the behaviour described above is not what a physical ESP32 would do (see also the linked thread).

jonschz, 2023-05-28:
#460 might be related

miso-mosi, 2023-06-06:
Here is the capture from the Wokwi simulation

Notes: The PWM frequency is 100 Hz, whereas, in the physical ESP32 board, it is 1 kHz. 

[image]

urish, 2023-06-06:
Fixed! Can you please test again and report?

miso-mosi, 2023-06-06:
The 4095 and the 4096 value gives 100% PWM now.. It matches the physical board observations made.  ✅
[image]

One observation: Just before the PWM starts, there is a different behavior in both ESP32 hardware and ESP32 Wokwi part.

This is observed every time the PWM switches from one 4093 to 4095, for example. 🔎
 
[image]
[image]

With real ESP32, there is a delay before the other pulses follow. There is a delay between the first pulse and the subsequent pulses.

urish, 2023-06-06:
Thanks, so we can close this issue! Can you please open a separate issue for the mismatch in the delays?

miso-mosi, 2023-06-06:
just an additional note

The timing is also fine now ( 1 kHz PWM wave) on both Wokwi and hardware ESP32
[image]

jonschz, 2023-06-06:
Thanks for the hardware tests and the fix!

I was surprised to find that `analogWrite(pin, 4095)` results in 100% power output instead of 99.975%,  so I looked into it and found that `arduino-esp32` maps the range `0..4095` to `0..4096` before passing the value down to the hardware:

https://github.com/espressif/arduino-esp32/blob/72c41d09538663ebef80d29eb986cd5bc3395c2d/cores/esp32/esp32-hal-ledc.c#L89-L94C1

I modified the wokwi example linked above to call use lower level functions. Everything works as expected - an internal duty value of `4095` results in almost 100 % output, while an internal duty value of `4096` corresponds to 100% on.

[image]

Dlloydev, 2023-06-22:
### Using the analogWrite function (8-bit pwm):

[Using Arduino analogWrite() on an AVR](https://wokwi.com/projects/368195060558294017) differs from using Arduino analogWrite() on an ESP32. 
* The results from an AVR simulation for duty at 256 shows continuously low, while in real hardware (Leonardo) there's a 4us pulse.  This pulse doesn't show up for duty = 0 because in analog.c they use digitalWrite to make it LOW. Similarily, for duty = 255, they use digitalWrite to make the output HIGH. Values higher than 255 aren't corrected.

[Using the ESP32](https://wokwi.com/projects/368197061730029569), both real hardware and simulation tests differ from the AVR analogWrite implementation.
* For duty = 256, (both hardware and simulation tests) the output is HIGH.
* For duty = 257, simulation results have the output HIGH, whereas on real hardware the output reverts to the same pulse width as duty = 1.

**Hardware results:**

[image]