wokwi/wokwi-features #588: I2C address of SSD1306 OLED display is incorrect

dhebbeker, 2023-06-04:
**Describe the bug**
The default I2C address of the Adafruit 128x64 screen is `0x3D`.

[image]

This does not match with

- the [Wokwi documentation](https://docs.wokwi.com/parts/board-ssd1306)
- the Wokwi implementation (if address is not `0x3C`, display does not work)

**To Reproduce**
 - I took the example code and changed the address: [Modified project](https://wokwi.com/projects/366623971276246017).

Output is:

```
Traceback (most recent call last):
  File "main.py", line 9, in <module>
  File "ssd1306.py", line 116, in __init__
  File "ssd1306.py", line 36, in __init__
  File "ssd1306.py", line 61, in init_display
  File "ssd1306.py", line 121, in write_cmd
OSError: [Errno 19] ENODEV
MicroPython v1.19.1 on 2022-06-18; ESP32 module with ESP32
Type "help()" for more information.
>>> 
```
**Expected behavior**
"Hello Wokwi" is written to the display

urish, 2023-06-05:
Thanks for taking the time to open this issue!

The module we are simulating is not the Adafruit one, rather the common model you can find on Amazon and Aliexpress, e.g.:

https://www.amazon.com/HiLetgo-Serial-128X64-Display-Color/dp/B06XRBTBTB. It looks like 0x3c is a commonly used address for this modules. I hope this clarification helps!

closed 2023-06-05 by urish

dhebbeker, 2023-06-05:
Thank you for the reply!

Just in case somebody else stumbles upon this: The default I2C address can be adjusted in Wokwi (great!):

```JSON
    {
      "id": "ssd1306",
      "type": "wokwi-ssd1306",
      "left": 100,
      "top": 230,
      "attrs": { "i2c-address": "0x3D" }
    }
```
Seen in the [Adafruit SSD1306 Showcase](https://wokwi.com/projects/344892392214626898).

urish, 2023-06-11:
You're welcome! Also added a note about this to the documentation