wokwi/wokwi-features #589: To extend Web API to allow downloading project files by name programmatically

shaos, 2023-06-05:
**Is your feature request related to a problem? Please describe.**
For now we can use Wokwi API to download diagram.json and to get Verilog representation of the circuit.
I believe that ability to download any file from the project will help to grow Wokwi ecosystem.

**Describe the solution you'd like**
To get diagram.json now we can use this API call:
https://wokwi.com/api/projects/366564630633368577/diagram.json
Similarly other files should be available as well:
https://wokwi.com/api/projects/366564630633368577/cgate.chip.json
https://wokwi.com/api/projects/366564630633368577/cgate.chip.v
https://wokwi.com/api/projects/366564630633368577/sketch.ino
and some kind of manifest listing all available files:
https://wokwi.com/api/projects/366564630633368577/manifest

shaos, 2023-06-15:
Not open for vote yet? ;)

urish, 2023-06-15:
Sorry, missed this one. Fixed it now!

urish, 2023-06-15:
https://wokwi.com/features?sort=newest#feature-1741447501

shaos, 2025-03-16:
Now it's possible to download ZIP-archive using API:

https://wokwi.com/api/projects/366564630633368577/zip

I think it's good enough for me ;)

closed 2025-03-16 by shaos

shaos, 2025-03-23:
it's closed, but I still can vote for it here https://wokwi.com/features