wokwi/wokwi-features #594: switch loop won't work within while or do-while loop

AlexanderTonn, 2023-06-09:
**Describe the bug**
Do-while or while loop does not ensure that the switch loop can be used multiple times within an iteration. 

**To Reproduce**
I've tried several things. I also tried to use  only while loops

Example:
iCase==1 and byValue==11
* the code is going into case 1, then into the else statement and iCase is getting iCase=10
* because of the break statement the switch loop will left
* the code is remaining only in the do while loop but is not going into the switch loop although the code should must going into Case 10 

**Expected behavior**
I've made a kind of setup wizard for setting up the etc module. The User have to insert by Serialinput an Option value and the value which have to be changed. For this I made at the end a Switch loop for detecting whether the inserted value was wrong or whether the user want to insert furthermore rtc settings. 

**Environment (please complete the following information):**
 - OS: macOS ARM M1
 - Browser Safari
 - Version 16.5
 - in Wokwi: MEGA2560

**Additional context**
```C++
// ... 
do{
              switch(iCase){
                // Set Year
                case 1:
                if(byValue < 00 || byValue > 99){
                  iCase = 100;
                }  
                else{
                  rtcDs3231.setYear(byValue);
                  Serial.print("RTC Year has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // Set Month
                case 2:
                if(byValue < 1 || byValue > 12){
                  iCase = 100;
                }else{
                  rtcDs3231.setMonth(byValue);
                  Serial.print("RTC Month has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // Set Date (Day of Month)
                case 3:
                byte byMonth = rtcDs3231.getMonth(xCentury);
                if(byValue < 1 || byValue > 31){
                  iCase = 100;
                }else if(byMonth == 1 && byValue <= 31){
                  rtcDs3231.setDate(byValue);
                  Serial.print("RTC Date has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // Set Day of Week
                case 4:
                if(byValue < 1 || byValue > 7){
                  iCase = 100;
                }else{
                  rtcDs3231.setDoW(byValue);
                  Serial.print("RTC Day of Week has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // Set Hour
                case 5:
                if(byValue < 0 || byValue > 23){
                  iCase = 100;
                }else{
                  rtcDs3231.setHour(byValue);
                  Serial.print("RTC Hour  has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // Set Minutes
                case 6:
                if(byValue < 0 || byValue > 59){
                  iCase = 100;
                }else{
                  rtcDs3231.setMinute(byValue);
                  Serial.print("RTC Minutes has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // set seconds
                case 7:
                if(byValue < 0 || byValue >59){
                  iCase = 100;
                }else{
                  rtcDs3231.setSecond(byValue);
                  Serial.print("RTC Seconds has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                case 8:
                // set Clockmode
                if(byValue < 0 || byValue > 1){
                  iCase = 100;
                }else{
                  rtcDs3231.setClockMode(byValue);
                  Serial.print("RTC 12h Mode has been set to: ");
                  Serial.println(byValue);
                  iCase = 10;
                }
                break;

                // Cancel the Setup routine
                case 9:
                xSetupActive = false;
                break;

                // Option change, want change more?
                case 10: 
                rtcLib.now(); 
                Serial.println("The RTC was set on: ");
                Serial.print(rtcDs3231.getYear());
                Serial.print("/");
                Serial.print(rtcDs3231.getMonth(xCentury));
                Serial.print("/");
                Serial.println(rtcDs3231.getDate());
                Serial.print("Current Day of Week: ");
                Serial.print(rtcDs3231.getDoW());
                Serial.print(rtcDs3231.getHour(xH12,xhPM));
                Serial.print(" : ");
                Serial.print(rtcDs3231.getMinute());
                Serial.print(" : ");
                Serial.println(rtcDs3231.getSecond());

                Serial.print("Do you want change more option?");
                Serial.println("(Y/N)");

                while (Serial.available() <= 0)
                ;

                if (Serial.available() > 0){
                  char cYesNo = Serial.read();
                  if(cYesNo == 'N' || cYesNo == 'n')
                    xSetupActive = false;
                }
                iCase = 0;  
                break;

                case 100:
                  Serial.println("Invalid value input");
                  Serial.println("Please consider the syntax: <option>,<value>");
                break;

              }
            }while(iCase==10);
```

AlexanderTonn, 2023-06-10:
I'm afraid that I have to retract my bug report. I followed your recommendation (why didn't I have this idea before? =D). The compiler on Arduino is doing the same as wokwi. So I'm only wondering why the compiler have obviously problems with switch loops within do-while or while-loops. Nevertheless I replaced the Switch loop against if-else loops and now it works as expected. I'm sorry for the useless bug report. I thought that this manner is not a normal. And probably it is an anomaly but not caused by wokwi.

closed 2023-06-10 by AlexanderTonn

urish, 2023-06-10:
Thanks for checking!

If I remember correctly, the Arduino GCC compiler doesn't handle new variables inside `switch` cases very well. A workaround would be to move the definition of `byMonth` out of the switch statement.