wokwi/wokwi-features #596: Application error: a client-side exception after browser window got resized too small

shaos, 2023-06-11:
**Describe the bug**
After browser window with opened Wokwi editor resized to extremely small size and resized back it's getting Application error when user clicks to editor tab:
"Application error: a client-side exception has occurred (see the browser console for more information)."
Console shows:
```
Error: Model is disposed!
    _assertNotDisposed https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:675
    isDominatedByLongLines https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:677
    _attachModel https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:690
    setModel https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:690
    NextJS 60
framework-ce84985cd166733a.js:9:69816
Error: Model is disposed!
    _assertNotDisposed https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:675
    isDominatedByLongLines https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:677
    _attachModel https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:690
    setModel https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js:690
    NextJS 60
main-51d231f41fd8d54e.js:1:8792
A client-side exception has occurred, see here for more info: https://nextjs.org/docs/messages/client-side-exception-occurred main-51d231f41fd8d54e.js:1:8809
Source map error: Error: request failed with status 404
Resource URL: https://wokwi.com/cdn/monaco-editor/0.33.0/vs/editor/editor.main.js
Source Map URL: ../../../min-maps/vs/editor/editor.main.js.map
```

**To Reproduce**
Open Wokwi circuit, change something in the circuit or diagram.json directly, don't Save, start gradually reducing window size until text editor disappears then open up the window back to normal resolution and click to diagram.json tab (see video)

https://github.com/wokwi/wokwi-features/assets/236968/9c3d4095-2957-4c49-846b-622af8f92521

**Expected behavior**
App should not mess with tabs and should not crash

**Environment (please complete the following information):**
 - OS: Linux
 - Browser firefox and chromium
 - Version 102.10.0esr and 90.0.4430.212-1~deb10u1

**Additional context**
Found it when Firefox window with Wokwi was maximized and I switched to DOSBox app where chose full screen mode that changed screen resolution to 640x480 - after coming back Wokwi tabs were changed to default and attempt to click to modified tab crashed the web-app. The same behavior is observed with other resolution changing apps as old full-screen DirectDraw or OpenGL windows apps running through Wine.

urish, 2023-06-11:
Thanks for reporting! I've applied a workaround, can you please test again now?

shaos, 2023-06-11:
Now it is not showing exception, but tabs are still switched to 1st one

urish, 2023-06-11:
Thanks! I guess we can live with that :)

closed 2023-06-11 by urish

shaos, 2023-06-11:
ok