wokwi/wokwi-features #600: Ability to change the MAC address of the simulated ESP32

Tuaghmohammad, 2023-06-17:
**Describe the bug**
 I created a Vscode project on two different folders and tried the same Wifi application, but it gives the same IP address of 10.13.37.2 in both, how can I define a different IP address for each ESP32. (The MAC address mybe same for every ESP32)

**To Reproduce**
[Steps to reproduce the behavior, including link to a simulation project where we can see the problem.](https://wokwi.com/projects/320964045035274834)

**Expected behavior**
I was expecting a network connection between two different ESP32 devices, each with a different IP address.

**Environment (please complete the following information):**
 - OS: [e.g. iOS/Android/Windows]: Windows 10
 - Browser [e.g. chrome, safari]: Google Chrome
 - Version [e.g. 22]: Chrome 114.0.5735.133

**Additional context**
I will create a wifi network with several ESP32 devices and do security tests between them and see the results with wireshark or similar tools and prepare an article accordingly.

urish, 2023-06-17:
Done!

You can now set the mac address by specifying a "macAddress" attribute for the ESP32 board in diagram.json. Here's an example:

https://wokwi.com/projects/367800078553330689

closed 2023-06-17 by urish

Tuaghmohammad, 2023-06-17:
Hi @urish, Thank you very much for your very fast response. 👍