wokwi/wokwi-features #602: Delay not working as expected

SergioGasquez, 2023-06-22:
Delays are not working as expected, setting 1-second delay results in about 2 seconds
## How to reproduce
Simulate a blinky example, e.g: https://wokwi.com/projects/341952826535051858

## Expected behavior
On real hw, the delay last, more or less, 1 seccond
[image]

urish, 2023-06-22:
Initial investigation shows that this is related to the ESP32 ROM accelerator (a mechanism that skips the ROM / bootloader and executes the user code directly). A quick workaround is to disable the ESP32 ROM accelerator by adding a `"fullBoot": "1"` attribute:

```json
        {
            "type": "wokwi-esp32-devkit-v1",
            "id": "esp",
            "top": 0,
            "left": 0.67,
            "attrs": {
                "builder": "rust-nostd-esp",
                "fullBoot": "1"
            }
        },
```

SergioGasquez, 2023-06-22:
The issue also appears in
- [S2](https://wokwi.com/projects/341955297469792851)
- [S3](https://wokwi.com/projects/349568945405035090)

C3 seems to be working fine

urish, 2023-06-22:
With ESP32-S2, it seems like same issue (ESP32 ROM accelerator).

For ESP32-S3/C3/C6 (and future chips), we don't use the ESP32 ROM accelerator. For ESP32-S3, it seems like there's an issue with the clock simulation, I'll investigate. The delay takes 4 seconds instead of just 1.

urish, 2023-06-22:
Update: ESP32-S3 clock issue found and fixed