wokwi/wokwi-features #605: Make changes to files to the files on SD card persistent between executions of the emulator

thjorth, 2023-06-22:
**Is your feature request related to a problem? Please describe.**
Files that are stored on the SD card are not persisted between executions. I am building a mockup of an app that should initialize and populate a file if it doesn't exist on the card. Then the hardware would allow the user to modify the data in the file. It would be lovely if the files that are created onto the SD card be persistent. And it would also be great to have a way of inspecting their contents or downloading them.

**Describe the solution you'd like**
I would like files that a program creates on the SD card to be persistent. Modifications to files on the SD card should also be persistent. And I would like an option to download those files or inspect their contents.

urish, 2023-06-22:
Thanks for opening this request, I opened it for voting!

Could Wokwi for VS Code also work for you? It might make more sense to implement this feature there.

thjorth, 2023-06-23:
Yes, it could also work if it was only implemented for VS Code.

BR Thomas

Den tors. 22. jun. 2023 kl. 21.05 skrev Uri Shaked ***@***.***

urish, 2026-10-02:
This is now implemented both in Wokwi for VS Code and wokwi-cli, using the `writeback` setting: https://docs.wokwi.com/vscode/project-config#sd-card-contents

closed 2026-10-02 by urish