wokwi/wokwi-features #608: Misleading dialog box on"save"

alto777, 2023-06-28:
It's not a bug, it's a feature.

When I say "save", there is a dialog pops up saying

"Leave page? Changes you made may not be saved."

<br>

TBH there is still something not intuitive or maybe even squirrely around saving, copying and dealing with projects in general. I have lost work because I am too old or dumb to understand what had been simple since 1984.

But the "leave page?" thing I usually see if I am leaving work to be destroyed if I do leave the page. I know what that means and what to do.

The dialog is spurious and should be changed or eliminated if there is nothing to say.

a7

urish, 2023-07-02:
Thanks for reporting!

urish, 2023-07-15:
Fixed

closed 2023-07-15 by urish