wokwi/wokwi-features #612: ESP32 Micropython always shows text in serial monitor and runs slower

rrp3168, 2023-07-05:
Lately, everytime I run ESP32 on Micropython, the serial monitor opens and displays some running instructions and the simulator runs slower.
This just started recently, like a couple of weeks ago.
Since then, the simulator runs slower, always under 50%.
Is there anyway to fix this?
The instructions shown are:
[image]

urish, 2023-07-05:
Hello, thanks for reporting!

The output that you are seeing is the ESP32 boot messages. We used to skip those - but it caused a bunch of other issues (example: #602). It might be possible to hide these messages - but for now, we always display them.

As for the simulation running slower - I'm not sure if it's related. Do you have an example project that used to run faster so I can look into it?

rrp3168, 2023-07-05:
Hello Uri:
Thank you so much for such a quick response and for clarifying the issue on the messages.
As for the low running time I thought it was relayed since I always had a high running percentage, between 90 and 100%.
But lately it is around or under 50%.
[image]
I am adding a link to a very simple circuit which runs slow. [https://wokwi.com/projects/369255077291936769](url)
If you see anything strange or have an idea of what could be the reason, please let me know.
Thank you so much for your time.
Best regards.

urish, 2023-07-08:
Thanks for the link! I just tested it on my laptop, and it seems to run at around 90-100%.

Which web browser are you using? Chrome usually works best, other browsers may be slower.

also, try placing `delay(0.01)` in the main while loop, this can considerably speed up the simulation.

rrp3168, 2023-07-08:
That was it!
I was using Mozilla Firefox. Switched to Chrome and it went to 100%.
Thank you very much!

closed 2023-07-08 by urish