wokwi/wokwi-features #618: Support for Particle Devices

chipmc, 2023-07-11:
I would like to see support for the Particle devices using deviceOS.  Particle is a significant IoT platform with a mature and well documented API and libraries.  Adding Wowki support could help is large community collaborate on IoT solutions without requiring everyone to purchase specific sensors or dedicate hardware to testing.

urish, 2023-07-28:
Thanks for the suggestion! 

Unfortunately, adding support for those devices will require a huge amount of effort. So unless particle is interested in sponsoring the development, or there's a big commercial customer interested in these devices, it is not likely to happen any time soon.

closed 2023-07-28 by urish

chipmc, 2023-07-28:
Understand, thank you.