wokwi/wokwi-features #620: esp-idf-hal UART inconsistent behavior

apollolabsdev, 2023-07-17:
Running the uart loopback example from the esp-idf-hal repo (https://github.com/esp-rs/esp-idf-hal/blob/master/examples/uart_loopback.rs) on the ESP32C3, the code seems to get stuck at the blocking read statement as the following println! statement does not get executed. When downloading the same code on real hardware works no issues are observed.

This is a link to the example project:
https://wokwi.com/projects/370479605507228673
Line 30 is where the execution halts.

urish, 2023-07-18:
Thanks for reporting! Pausing the simulator shows that the pins (5, 6) do not get configured correctly as UART1 RX/TX:

[image]

Investigating why

urish, 2023-07-18:
Actually, scratch that - I paused the simulation too early. For some reason, it takes the code more than 1 second to reach the actual UART test code, and I hit the "pause" button to early. So the pins do get assigned correctly:

[image]

apollolabsdev, 2023-07-18:
I encountered the same issue as I was trying to debug. Looking at the console, it seems that the device is still booting before the pins get finally configured.

closed 2023-07-18 by urish

apollolabsdev, 2023-07-18:
Check it out. Working as expected now, thanks!