wokwi/wokwi-features #624: Give uninitialized variables random values

nosknut, 2023-07-20:
In a real microcontroller, an uninitialized variable could have any value based on what was previously in memory. It would be useful to consistently demonstrate this with wokwi by having uninitialized variables start off with randomized data.

sutaburosu, 2023-07-20:
This isn't the case, at for least AVR microcontrollers. Uninitialised variables will be cleared to 0 by libc at startup, except those in [the .noinit section](https://www.nongnu.org/avr-libc/user-manual/mem_sections.html). See [this sketch](https://wokwi.com/projects/330312379574256210) for an example of a .noinit variable.

On ESP32 it seems you would use the [`__NOINIT_ATTR`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/memory-types.html#noinit-dram) to get the same behaviour.

urish, 2023-07-28:
Well explained!

closed 2023-07-28 by urish