wokwi/wokwi-features #626: Display offset 55 pixels up

Ritbulau, 2023-07-25:
**Describe the bug**

When trying to launch my project, the display ( SSD1306 OLED display ) on the screen shifts 55 pixels up, which should not happen and does not happen on a real board and in debug mode. At the same time, in some cases, the display does not just shift, but also, as it were, begins to display different parts of what should be displayed on the screen from above.

**To Reproduce**

Just run the project and the bug will appear.
To understand how everything should look, run the debug mode. To see a bug with scrolling parts of what should be on the screen from the top of the display, press the Fire button (it is located in the center of the "control panel")

Project link:
https://wokwi.com/projects/370145885662336001

**Expected behavior**

I expext to see the screen, how it looks in debug mode. 
I tested the program on a real arduino nano, and everything looks fine on it (as in debug mode).

**Environment (please complete the following information):**
 - OS: Windows
 - All browsers (Chrome, Edge, Firefox etc.)

**Additional context**

This is my chess project based on the Doom Nano engine (in which this problem is not observed), and initially I ignored this problem, just always running debug mode, but then, at one point, I added a couple of lines of code fixing one bug in the game, and debug mode started giving the error "Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error during build: text section exceeds available space in board", which does not happen during normal startup, as well as after testing the program on a real board, there is no such problem either, and the sketch takes 22866 bytes (74%) device memory and 1831 bytes (89%) of dynamic memory. For a more convenient understanding of the problem, I temporarily removed this part of the code so that there would be no such error, so that you could start the debug mode, but I will leave a link to the project with the current version of the code with this error ( https://wokwi.com/projects/371239309846964225 ). I won't complain about this problem so much, because the debug mode is still in beta, but I will be glad if you fix at least one of these two problems so that I can continue developing the project. I do this, including with the help of your emulator, because with its help I can record the gameplay of a video game in good quality directly from the display, rather than recording the real display with the camera.

urish, 2023-07-25:
Thanks for reporting! Unfortunately, I won't have time to look into it over the next few weeks.

However, if someone else would like to try, I just open sourced the SSD1306 simulation model in the factor of a custom chip:

https://wokwi.com/projects/371050937178768385

(see ssd1306.chip.c for the implementation).

Regarding the out-of-memory issue when debugging: I think the easiest solution would be to switch to the Arduino Mega while debugging. Compiling in debug mode does increase the size of the code, as the compiler is no longer able to do some of the optimizations, so more memory is required.

urish, 2026-02-02:
Tested again, seems like this is now working as expected.

P.s.