wokwi/wokwi-features #627: DS1307 Reset State Adjustment

apollolabsdev, 2023-07-26:
When working with the DS1307 RTC it seems the reset state doesn't necessarily match the datasheet description. While this is not a huge deal as one can update the device with desired values, it would be nice to align with the datasheet to avoid confusion.

Quote from Datasheet
> On first application of power to the device the time and date registers are typically reset to 01/01/00 01 00:00:00 (MM/DD/YY DOW HH:MM:SS)

https://eu.mouser.com/ds/2/256/DS1307-99500.pdf

urish, 2023-07-26:
Thanks for letting me know about this! 

The DS1307 is [initialized with the current system time](https://docs.wokwi.com/parts/wokwi-ds1307#simulation-behavior). Since the value isn't persisted across simulator runs, if we initialized it every time with zeros, users would always have to manually set the time.

There is an open issue to let users set the initial value - #300. I think one solution, that would also be simpler, would be to have an diagram.json attribute that determines whether the time is initialized as the datasheet says or to the current time. WDYT?

apollolabsdev, 2023-07-27:
It seems I overlooked the description in the Wokwi documentation, well, and the issue 😃 However, yes, I agree with your suggestion as seems like a good way to go. It allows for changing the initial time without necessarily writing the code for it in all projects. Thanks!

closed 2023-07-27 by apollolabsdev

urish, 2023-07-28:
Update: we now have an "initTime" attribute that allows you to set the initial time. Set it to "0" to get the same behavior as in the datasheet.

https://docs.wokwi.com/parts/wokwi-ds1307#attributes