wokwi/wokwi-features #632: Allow text labels to be placed near LED (or other) objects

rpbgit, 2023-08-10:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

rpbgit, 2023-08-10:
add a "Text" object that can be dropped into the "wiring" diagram.  allow vertical and horizontal orientation and configurable text font size.  allows one to keep things sorted if (for example) you have many Green LED's...

urish, 2023-08-10:
Hello! For LEDs specifically, there's a label attribute which allows you to add a short description text next to the LED: https://docs.wokwi.com/parts/wokwi-led#attributes

A similar attribute is also available for pushbuttons.

For other parts, there's an open proposal for a text element, see #43 for details + example. You can also vote for it at https://wokwi.com/features#feature-782100051.

closed 2023-08-10 by urish

rpbgit, 2023-08-11:
Hi Uri,
thx for the reply.   I will look at using the label attribute in the .json screen.  what i was thinking more along the lines of was being able to do so from the simulator screen directly so knowledge of the json isnt needed/necessary for someone only interested in the code and simulation, not how the sim is implemented.  i realize you are using it as a youtube teaching vehicle for javascript and other purposes.

urish, 2023-08-11:
Yeah, once we finalize the work on it we plan to also create user interface for this feature