wokwi/wokwi-features #633: Blue pill STM32F103C8 ADC not fully working

dzid26, 2023-08-17:
**Describe the bug**
ADC_FLAG_EOC never gets set

**To Reproduce**
Steps to reproduce the behavior, including link to a simulation project where we can see the problem.
1. clone https://github.com/dzid26/StepperServoCAN/tree/wokwi/firmware
2. install platmformio
3. build default firmware (ServoCAN_release)
4. start wokwi with debugging
5. Revert automatic changes in `launch.json` done by Platformio. Check if the path to `arm-none-eabi-gdb` works
6. start debugger
7. press pause in vscode debugger

**Behavior**
The program gets stuck checking the state [ADC_FLAG_EOC ](https://github.com/dzid26/StepperServoCAN/blob/wokwi/firmware/src/BSP/board.c#L324)

Wokwi shall emulate the behavior of the peripherals' registers. 

**Environment (please complete the following information):**
Windows11
VSCode extension

urish, 2023-08-18:
Thanks for reporting!

Yeah, the Blue Pill STM32F103 is not fully supported yet. What peripherals do you use other than the ADC there?

dzid26, 2023-08-18:
Bluepill is still super popular. Would be nice if it was officially supported. 

I am using ADC, SPI, TIM., RTC and CAN. 
Is SPI implemented similar to [l031k6](https://docs.wokwi.com/parts/board-st-nucleo-l031k6)? 

I was thinking to use wokwi-cli for some automated testing. I started using [Unity](https://docs.platformio.org/en/stable/advanced/unit-testing/frameworks/unity.html) within Platformio testing framework. Platformio automatically uploads test firmware and then can communicate using [semihosting](https://github.com/dzid26/StepperServoCAN/blob/wokwi/firmware/platformio.ini#L109-L122) to get results. This works pretty great.  
I actually tried quickly to do something similar [with wokwi-cli](https://github.com/dzid26/StepperServoCAN/blob/wokwi/firmware/platformio.ini#L135C15-L156) and [USART](https://github.com/dzid26/StepperServoCAN/blob/wokwi/firmware/test/unity_config.c#L29-L68), but didn't get any output in CLI. Maybe USART is not implemented?

Another great use case would be to call wokwi directly from Platformio using some custom [upload](https://docs.platformio.org/en/latest/projectconf/sections/env/options/upload/upload_protocol.html)/[debug_tool](https://docs.platformio.org/en/latest/projectconf/sections/env/options/debug/debug_tool.html). It seems like it should be doable with wokwi-cli. Even though it doesn't provide GUI, it could be possible to quickly run a debugger.

urish, 2023-08-18:
Thanks for the super detailed response. 

Here's the current status:
- RCC implemented
- GPIO implemented
- USART is implemented, but pin mapping is not fully working on the bluepill, and that's probably the reason you are not getting any serial output.
- All other peripherals (ADC, SPI, CAN, DMA, I2C, CRC, RTC, IWDG, PWR, WWDG, USB) are not implemented yet.

What would be the minimum set of peripherals that would be useful for you to be able to test your project?

dzid26, 2023-08-19:
Thanks for the clarification.

ADC would be nice to replicate my existing unit test. 
For unit testing with Unity, some sort of printing to outside world would be required. Either via USART or perhaps [SWI](https://developer.arm.com/documentation/dui0058/d/semihosting/semihosting/the-swi-interface)! This is what I am using currently on actual hardware with [OpenOCD](https://interrupt.memfault.com/blog/arm-semihosting) GDB server redirecting to console. It would be nice if wokwi could print via SYS_WRITE 0x5 directly to GDB or directly to the console.

Regarding trying to emulate my main program (not the unit test), I would need again ADC and SPI to pass my initialization without commenting stuff out. And then TIM-triggered interrupts to actually run foreground tasks. 
Are SysTick interrupts supported? I may switch from TIM to SysTick timebase.

urish, 2023-08-19:
Yes, SysTick is indeed supported.

I'll take a look at what it would take to get USART to work so you can get started with unity. 

How would you prioritize SWI vs ADC/SPI?

SysTick should work.

urish, 2023-08-20:
Update: USART output should be working now

dzid26, 2023-08-20:
> Update: USART output should be working now

I did some tests with USART.  I think printing only works for USART1 at A9&A10 pins?  Initially, I tried remapping to PB&PB7 and also USART2 and printing didn't work. 

I noticed that UART RX pin needs to be pulled up (not floating), to appear when pausing the simulation. 
[image]
Incidentally, pins A3...A6 (red circle) are simple digital inputs with internal pull-up (`GPIO_Mode_IPU`) and simulator thinks they have some special functions. 

> How would you prioritize SWI vs ADC/SPI?

Since I have some basic USART to play with, I would say ADC, SPI,  SWI.

urish, 2023-08-20:
> I noticed that UART RX pin needs to be pulled up (not floating), to appear when pausing the simulation.

Yeah, modeling the behavior correctly will require some non trivial amount of work, so remapping isn't there yet, and it has this strange behavior with the pull-ups. I hope to get to fix this at some point. But for now, I'll prioritize getting ADC and then SPI working.

Glad to read the unity test passed! One step forward :)

What are your plans going forward with this?

dzid26, 2023-08-20:
> What are your plans going forward with this?

I didn't expect we would even get this far :) 

If ADC works, I would mock up the diagram for the Unity test in the CI. 

Then with SPI (+TIM), I could start thinking of creating a simple [plant model](https://www.manmonthly.com.au/plant-modelling-a-first-step-to-early-verification-of-control-systems/) using [custom chip](https://www.youtube.com/watch?v=yzdCS3A4DvU). Technically, I don't have to do it, since I test on hardware anyway. But I want to give it a try to see what Wokwi can do!

urish, 2023-08-21:
Alright, I'll ping here once there's some progress with the ADC. Can you please open another issue for SPI?

dzid26, 2023-08-22:
Thanks.

I created SPI issue. https://github.com/wokwi/wokwi-features/issues/637

urish, 2023-09-02:
Update: ADC implemented. Example: https://wokwi.com/projects/374772127111874561

Missing features:
- Scan mode 
- Injected channels 
- Discontinuous mode 
- External triggers 
- DMA 
- Dual ADC Mode 
- ADC2 

Please share your feedback

urish, 2026-01-17:
We now have a documentation page with an up-to-date status of STM32F103C8 peripherals: https://docs.wokwi.com/parts/board-stm32-bluepill

dzid26, 2026-01-17:
I think it would be clearer if the doc said: "Dual ADC mode missing". Instead of "Dual ADC/ADC2 are missing"
If i understand correctly.