wokwi/wokwi-features #639: More clear erradic behaviour for buttons when missing pullup/pulldown circuit

nosknut, 2023-08-24:
It would be nice if buttons produced more confusing inputs for microcontrollers when they lack a pullup/pulldown circuit.
This would make it easier to debug.

In the case below the value does semi-randomly change when you click the button,
however it would be more indicative of a problem if it were to flicker faster
without any interaction (Like the bounce?).

https://wokwi.com/projects/373955312754164737

urish, 2023-08-27:
Thanks, I opened this for voting. In real life, there are many aspects which can affect floating pins (e.g. touching them with a finger, mains switching noise, etc.), so I'm not sure what would be the best way to model this. 

Would you mind recording a few samples on a physical Arduino board to see if you can observe any pattern?

urish, 2024-06-30:
Closing this issue as it hasn't received any votes over than past 6 months.

closed 2024-06-30 by urish