wokwi/wokwi-features #640: Oscilloscope

ME-TECH-ELECTRONICS, 2023-08-28:
Every time I start the simulation I get a problem like whether the microcontroller is giving a output signal or not. When I tried other simulation software I got the oscilloscope feature my would like to get this feature to added in wokwi

urish, 2023-08-29:
Hello! Have you tried the [logic analyzer](https://docs.wokwi.com/guides/logic-analyzer)?

ME-TECH-ELECTRONICS, 2023-08-30:
Yes I tried but it requires another piece of software to view the signal. But a oscilloscope i mentioned here can show the realtime signal in wokwi without any other software.

urish, 2023-08-30:
Something like this? 

* https://wokwi.com/projects/360935045930911745
* https://wokwi.com/projects/361237697368753153

drf5n, 2023-12-30:
The Wokwi-Chip-Scope is developed and documented at:

https://github.com/Dlloydev/Wokwi-Chip-Scope