wokwi/wokwi-features #641: Adding Files to a Project

brightproject, 2023-08-29:
@urish please relocate issue into https://github.com/wokwi/wokwi-features
I can't recover my Discord password - that's why I'm writing about problems here.
1. I simulate my project that runs on real hardware.

[image]

https://github.com/wokwi/wokwi-docs/assets/1788098/bb6748ee-96c4-4987-9396-8f1ecae9db82

Added 16 "*h" files and can't save project.
[image]
Is this a limitation of the free version or a bug?
I managed to save the project archive - help me figure out why it is not saved and why the buttons do not always work adequately.
[Flightinstruments_TFT_eSPI.zip](https://github.com/GyverLibs/EncButton/files/12448649/Flightinstruments_TFT_eSPI.zip)
The project is quite interesting, and can greatly popularize your simulator.
In case of full adaptation of the code to the simulator, you can add it to the section [featured](https://wokwi.com/esp32) projects.
Please see it if possible.
Any questions are welcome.
3. Strange work with the button library [EncButton](https://github.com/GyverLibs/EncButton), the code compiles every other time.
[Issues](https://github.com/GyverLibs/EncButton/issues/42) with internal pullups, does the simulator not simulate internal pullups well?
4. The display
`"type": "board-ili9341-cap-touch"`
displays the image shifted to the right, you can see it in the screenshot above.
I wanted to [try](https://wokwi.com/projects/343960939576427090) another debug board, but I could not connect a display `ili9341` with the [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI/) library, nor did I find examples - apparently it does not support these displays yet.
`"type": "board-esp32-s3-devkitc-1"`
As far as I understand, the simulator is not a 100% likeness of a real device, and you should not expect miracles from it.
There were thoughts to make a purchase, but the simulator did not work on the PC in [VS Code](https://docs.wokwi.com/vscode/getting-started), and there are still bugs in the online version.

GyverLibs, 2023-08-29:
@urish solved, thanks!

brightproject, 2023-08-30:
> 4\. I'm curious - what issues are you having with the VS Code version?

Thanks for the replies.
In `VS Code` there is no possibility to edit the scheme visually, only in the file `diagram.json`
The speed of simulation execution is still very far from the speed of execution on a real microcontroller.
I tried to increase the flash memory and sdram, so if they realized the opportunity to increase the `SRAM` - the real esp32 `SRAM` has 512 kB.
Therefore, graphic displays work very fast.
Even a simple example works for me on [stm32f401](https://wokwi.com/projects/374062283976666113)
And in the simulator on `esp32` slows down.
Perhaps you need to take `esp32-s3` for simulation?
Or does the simulation have one speed of operation for all microcontrollers?