wokwi/wokwi-features #643: rfid reader

prodc749, 2023-09-07:
include a rfid reader object based on the Mifare RC522 module that can be found [here](https://www.amazon.com/HiLetgo-RFID-Kit-Arduino-Raspberry/dp/B01CSTW0IA/ref=sr_1_5?crid=1PS0G1YK21LKM&keywords=mifare+rfid+module&qid=1694048179&sprefix=mifiare+rfid+modul%2Caps%2C483&sr=8-5)

there would also be a rfid card object that when moved infront of the reader simulates what would happen when you create the object it has a random hex but you can edit it in the simulation

urish, 2023-09-18:
Closing, this is a duplicate of #359

closed 2023-09-18 by urish

AbdennourBELKACEM, 2024-11-05:
please include a rfid reader object based on the Mifare RC522 module, then community is in need for this module in the simulator

djedu28, 2024-11-05:
@AbdennourBELKACEM

1. Before commenting, check previous comments.
- This issue was closed because it's a duplicate of https://github.com/wokwi/wokwi-features/issues/359

2. You can build your own CI Mifare RC522
- https://docs.wokwi.com/chips-api/getting-started

3. As an alternative to RC522, I made my own RDM6300 CHIP, you can use it

- In the past I had the need to emulate an RFID reader, and I did a basic simulation of the [RDM6300 Chip](https://github.com/djedu28/wokwi-rdm6300-custom-chip) <https://github.com/djedu28/wokwi-rdm6300-custom-chip>
[image]

## diagram.json
```json
{
  "version": 1,
  "author": "Edu Santos",
  "editor": "wokwi",
  "parts": [
    { "type": "wokwi-arduino-uno", "id": "uno1", "top": 0.6, "left": -0.6, "attrs": {} },
    { "type": "chip-rdm6300", "id": "chip1", "top": 97.02, "left": 302.4, "attrs": {} }
  ],
  "connections": [
    [ "chip1:VCC", "uno1:5V", "red", [ "h-28.8", "v115.2", "h-86.4" ] ],
    [ "chip1:GND", "uno1:GND.2", "black", [ "h-19.2", "v124.8", "h-115.2" ] ],
    [ "chip1:RX", "uno1:1", "gold", [ "v0", "h30.61", "v-153.6", "h-192" ] ],
    [ "chip1:TX", "uno1:0", "green", [ "h11.41", "v-124.8", "h-166.9" ] ]
  ],
  "dependencies": { "chip-rdm6300": "github:djedu28/wokwi-rdm6300-custom-chip@0.28.2" }
}
```
---- 

> 1. Antes de comentar, verifique os comentários anteriores.
> - Este issues foi fechado por ser uma duplicata de https://github.com/wokwi/wokwi-features/issues/359
> 
> 2. Você pode construir seu próprio CI Mifare RC522
> - https://docs.wokwi.com/chips-api/getting-started
> 
> 3. Como alternativa ao RC522, fiz o meu próprio CHIP RDM6300, você pode usá-lo
> 
> - No passado tive a necessidade de emular um leitor de RFID, e fiz uma simulação básica do [RDM6300 Chip ](https://github.com/djedu28/wokwi-rdm6300-custom-chip)  <https://github.com/djedu28/wokwi-rdm6300-custom-chip>
> [image]
>