wokwi/wokwi-features #644: Bug in ESP 32 DHT 22.

PAULisaac12, 2023-09-12:
I entered the WOKWI website for our project and encountered an issue with the ESP32 DHT22. Despite facing this challenge, we managed to overcome it and successfully completed our new project. However, due to this issue, our initial expectations were not met, and I lost the progress of my project, which was the result of our hard work.

*Environment (please complete the following information):*
- OS: Windows 11
- Browser: Chrome
- Version: 4

*Additional context:*
We encountered a problem with the WOKWI DHT22. Upon investigation, I identified the issue with pins D4 and D5, which were connected to the humidity sensor. As a result, we were unable to submit our project. While learning about the WOKWI DHT22, I came across this type of problem, and unfortunately, it led to the loss of our project progress.

[image]
[](https://wokwi.com/projects/322410731508073042)
My details ,
         Email : [mr.paulisaac1212@gmail.com] 
Founded by : 
             PAUL ISAAC . S

urish, 2023-09-12:
Please take a look at the following example: https://wokwi.com/projects/322410731508073042

PAULisaac12, 2023-09-12:
In that link it has not sensing with the humidity sensor in D4 and D5. So check that issue. I has attached the screenshot below.
[image]

urish, 2023-09-12:
So it works with D15 but not D4/D5? Interesting

Have you tried it with D5 on real hardware?

PAULisaac12, 2023-09-12:
I'm not doing in reality even I am learner that's why I'm mailed about this

PAULisaac12, 2023-09-12:
I can try in hard ware and report it with the proof file thank you.

urish, 2023-09-12:
That would be very helpful

PAULisaac12, 2023-09-12:
but the report is confirm what will u do

urish, 2023-09-12:
If we know the behavior is different between the simulator and hardware we can start investigating why

PAULisaac12, 2023-09-12:
if the bug is confirmed and the problem founded by me any payment will be provided for the report.

urish, 2023-09-12:
I wish we were in a situation we could provide payment to anyone who submits a bug report. Unfortunately, we are not.

K1rdro, 2024-03-23:
I tested this on my hardware, and can confirm this is IRL behavior with this library. I was able to get it to return values after some tweaking of the code, but I would recommend using the available Adafruit DHT sensor library instead.

closed 2024-03-23 by urish