wokwi/wokwi-features #647: A4988 Driver Stepper Motor

skolodi, 2023-09-26:
A4988 Is the most popular Driver for Stepper Motor like Nema17.
  **### That Driver have to be on Wokwi !**
Vote for that improvement, please.

urish, 2023-09-27:
@Droog71 mind looking into this?

skolodi, 2023-09-27:
Sorry...can't  see your comment complete. May be the picture or the link not attached?

Droog71, 2023-10-12:
The only issue I can find is the reading from the first chip is not correct when there are multiple in the same project.
This version of the chip has attributes and the other one being used does not. My guess is it has something to do with attributes. I remember having a similar issue in the past and it had something to do with attributes. The other custom chip used here is very old and does not have demultiplexing  or attributes. It was modified by someone and the code is very different, so it is tough to compare the two.

Here is the project posted here with two chips.
The simulator does not crash.
I don't understand the comment about COM being connected to an Arduino pin instead of GND.
COM is the analog signal pin from the MUX.
EN works either way, connected to GND or an Arduino pin.

So, the only issue is the readings when two chips are in a project.
Not sure how to fix that just yet.
I will have to experiment.

@skolodi 
@urish

closed 2023-10-12 by urish

skolodi, 2023-10-18:
Hi again. I still have problem with the MUX cd74hc4067 #647.
May be other bugs, but with 4 MUX, 
I need:
All the 4 MUX with the same S0, S1, S2, S3 ( const byte MUX_ADDR [4] = {A3, A2, A1, A0};)
All the 4 MUX with the same SIG (const byte MUX_OUTPUT (A8);)
Each MUX with diferent Enable. Mux 4 on pin 25, Mux3 on 24, Mux2 on 23, Mux 1 on 22 (const byte MUX_SELECT [4] = {25, 24, 23, 22};)
But don't take the correct measures.

See the project with the problem here: https://wokwi.com/projects/376946283692116993
If I simulate it on Simulide, it works, but on Wokwi don't. 

Thanks for your help and time.

skolodi, 2023-10-18:
Hi again @urish. May be the change I made on the Example Project made by @Droog71 , shows the problem simplier and better. 
This works on real Arduino, works on Simulide, but I can't make it works fine on Wokwi.
All the 4 MUX with the same S0, S1, S2, S3 ( const byte MUX_ADDR [4] = {A3, A2, A1, A0};)
All the 4 MUX with the same SIG (const byte MUX_OUTPUT (A5);)
Each MUX with diferent Enable. Mux 4 on pin 5, Mux3 on 4, Mux2 on 3, Mux 1 on22 (const byte MUX_SELECT [4] = {5, 4, 3, 2};)
But don't take the correct measures of the potenciometers.

https://wokwi.com/projects/378946827241484289
Regards.

Droog71, 2023-12-08:
> Hi again @urish. May be the change I made on the Example Project made by @Droog71 , shows the problem simplier and better. This works on real Arduino, works on Simulide, but I can't make it works [...]

The problem is sharing the SIG pin.
I can make your project work by switching the SIG pin's mode to INPUT whenever the EN pin is high.
This will have to be done with a configurable attribute.
Otherwise, it will cause issues with other projects using a different configuration.