wokwi/wokwi-features #648: Pi Pico uart1 doesn't seem to be working

DrJackieBright, 2023-10-01:
**Describe the bug**
uart1 does not transmit or receive any data on pins gp4/5 or 8/9

**To Reproduce**
https://wokwi.com/projects/377341449400065025
Connect pins gp0 and gp1 (uart0) to gp5 and gp4 or gp9 and gp8 and send data between the two uarts

**Expected behavior**
Both uarts receive data from the other and print the result 

**Environment (please complete the following information):**
 - OS: Windows
 - Browser: Firefox
 - Version: 

**Additional context**
I verified that uart0 is working by connecting pins 0&1 together

urish, 2023-10-01:
Thanks for reporting, issue reproduced!

DrJackieBright, 2023-10-01:
I also just tried it on Android Firefox v118.1.0 and Android Chrome 117.0.5938.61 and got the same result

urish, 2023-10-01:
Found the issue, working on a fix

urish, 2023-10-01:
Fix is up! Can you please check now?

DrJackieBright, 2023-10-01:
Looks like it's working, thank you. This was driving me crazy thinking I had my project hooked up wrong

urish, 2023-10-01:
Thanks for checking! Most of the times, the opposite is actually true - users miss something and they blame the simulator. But in your case it was actually a simulator bug. While at it, I noticed that we also don't respect the UART baud rate setting for the RP2040, so a fix to that is also on the way.

closed 2023-10-01 by urish