wokwi/wokwi-features #649: Odd failure in perfect old project

alto777, 2023-10-02:
I was surprised to find that the least change to an old project broke it.

Visit

[This project](https://wokwi.com/projects/373410350749940737)

and cause it to rebuild, watch it break.

There should not have been any trouble; there was

I'm on MacOS 10.14.5 using Chrome Version 116.0.5845.187 (Official Build) (x86_64)

Adding the Adafruit_Neopixel library, which I feel I did have in there originally, then allowing a failed JSON parse to "retry", seems to fix it all up.

Guessing: I got no clue.

alto777

urish, 2023-10-02:
Thanks for reporting! The issue with "Retry" actually happens whenever you have any saved project with some wires but no libraries, and try to add a new library to that project. e.g.: https://wokwi.com/projects/377501999823647745 

Investigating.

urish, 2023-10-02:
The other issue, with the missing library, is related to a change in our build servers from a month ago - it doesn't install any libraries by default anymore. We ran an automated migration script, but it seems to have missed your project since it has a space between the "#" and the "include". If you want, I can run the migration script manually on your project to add a libraries.txt file (or you can do it if you have access to it).

urish, 2023-10-02:
Uploaded a fix for the "Retry" issue

alto777, 2023-10-03:
THX. Don't take any action on the other issue, with the missing library, on my account. I can role with anything that comes my way, I was just curious.

Also I guess I never noticed during the former days when some libraries were atou-included. I agree the idea to not do that is a good move.

a7

closed 2023-10-03 by urish