wokwi/wokwi-features #652: 运行指定的代码

TTORainyNight, 2023-10-09:
我在VSCode上配置了wokwi，并且指定micropython固件，在按下F1后，成功运行了esp32的仿真环境，并且实现了几条简单的命令。
但是esp32只能在终端交互模式下运行，我想知道如何让它运行指定的.py源代码，在VScode中。
[image]

rasca, 2023-10-15:
Hi, I just translated the issue to English and found I have the same problem.

Last version of vs code and the extension.

From VS Code when setting up a microPython project I don't get the main.py and boot.py executed. I'm using the bin and elf files from micropython.

My diagram.json:

```
{
    "version": 1,
    "author": "Iván Raskovsky",
    "editor": "wokwi",
    "parts": [
      {
        "type": "wokwi-esp32-devkit-v1",
        "id": "esp",
        "top": 0,
        "left": 0,
        "attrs": {}
      }
    ],
    "connections": [ [ "esp:TX0", "$serialMonitor:RX", "", [] ], [ "esp:RX0", "$serialMonitor:TX", "", [] ] ],
    "dependencies": {}
  }
```

My wokwi.toml:
```
[wokwi]
version = 1
firmware = './ESP32_GENERIC-20231005-v1.21.0.bin'
elf = './ESP32_GENERIC-20231005-v1.21.0.elf'
```

My main.py:

```
import machine
import time

pin = machine.Pin(2, machine.Pin.OUT)

print("Starting")

while True:
    pin.value(1)
    time.sleep(1)
    pin.value(0)
    time.sleep(1)
```

And an empty boot.py.

With this I cannot use the repl. But if I add `"attrs": {"env": "micropython-20231005-v1.21.0"}` to the diagrams.json I can use the repl but the main doesn't run and isn't in the filesystem.

@urish do you need any other info? or a pointer on where to look to fix this issue would be great. Thanks!

TTORainyNight, 2023-10-20:
Thank you for translating my issue. I apologize for my limited proficiency in English.

As you can see, this paragraph was translated by Chatgpt.

I have discovered that wokwi operates as a simulator in C language. It loads but does not run initially. In order to connect the C/C++ plugin to wokwi as a debugger, the configuration is set in the .vscode/launch.json file. The official documentation provides more information on this:

[Link to wokwi documentation on debugging](https://docs.wokwi.com/vscode/debugging)

On one hand, I would like to load the wokwi environment into the Python plugin. However, when selecting the Python environment in the plugin, I don't see wokwi listed. I only see my local Python interpreter, and I can't find the Python virtual interpreter provided by wokwi.

On the other hand, I would like to change the debugger to the Python plugin within wokwi, but I haven't found any relevant support. Specifically, I couldn't find the necessary launch.json support for micropython.

Currently, I am still unable to run my own .py files in VS Code and can only complete the process on the wokwi website.

urish, 2023-11-08:
Thanks for translating! No need for additional info - rather, time to spend on simplifying this / documenting it. 

In general, it is possible to use the virtual serial port over TCP to upload files into micropython, or create a firmware image the contains a file system with the binary files. However, both options are not straightforward and therefore I need to find time to sit down and write good documentation on how to do this.

rasca, 2023-11-09:
Thanks @urish , I'll be working on a related thing this weekend. Any pointers on how to do this would be much appreciated, I can then help with the documentation after I have a successful try.

Thanks!

urish, 2023-11-09:
There's an example setup here: https://github.com/minasouliman/wokwi_esp32_micropython

You can find more detailed by looking for past discussions about micropython + vscode on the [discord server](https://wokwi.com/discord)

urish, 2023-12-22:
Update: you can also use [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html) to execute scripts in micropython running inside Wokwi for VS Code. Here are the Instructions for Raspberry Pi Pico:

1. Download the micropython firwmare (e.g. [for the Pi Pico](https://micropython.org/resources/firmware/RPI_PICO-20231005-v1.21.0.uf2))
2. Configure wokwi.toml to point to the firmware and enable RFC2217 TCP port, e.g. 
```toml
[wokwi]
version = 1
elf = "RPI_PICO-20231005-v1.21.0.uf2"
firmware = "RPI_PICO-20231005-v1.21.0.uf2"
rfc2217ServerPort = 4000
```
3. Configure diagram.json for the target board (Pi Pico):
```json
{
  "version": 1,
  "author": "Uri Shaked",
  "editor": "wokwi",
  "parts": [
    { "type": "wokwi-pi-pico", "id": "pico" }
  ],
  "connections": []
}
```
4. Create a main.py script with your MicroPython code, e.g.
```python
from machine import Pin
from utime import sleep

print("Hello Wokwi!")

led = Pin(25, Pin.OUT)
while True:
  led.toggle()
  sleep(0.5)
```
5. Start the simulation, and keep the simulator tab visible (otherwise, the simulation will pause and mpremote may timeout)
6. Run the script with mpremote:
```
python -m mpremote connect port:rfc2217://localhost:4000 run main.py
```

Enjoy!

I only tested it on Pi Pico so far. Once we get it tested on additional boards, I'll create a demo repo and add these instructions to the docs.

urish, 2023-12-22:
Created a full example repo, with instructions and three examples: Pi Pico, ESP32 and ESP32-C3. 

https://github.com/wokwi/wokwi-vscode-micropython

Will also link to it from the docs.

closed 2023-12-22 by urish

Josverl, 2024-01-24:
or checkout : https://github.com/Josverl/vfs_merge

urish, 2024-01-25:
> I stalled working on it as wokwi gave too little feedback on errors in the source code , so i reverted back to testing on hardware and TDD.

Can you please elaborate on that? What difference are you seeing between Wokwi and the actual hardware?

Josverl, 2024-01-25:
Sure, and definitely not meant as a flame
 1. as I was essently trying to find the corrrect way to reposition and glue together two parts of an .uf2 file, I got it wrong more than once. at that point wokwi would just show me nothing, while on hardware I still got some feedback through the uart.
 I also find on esp32 that it takes a while for the wokwi uart to start working - perhaps its the same here.

 2. I struggled to find information on the memory that is assigned to the virtualized devices. and as most of the vfs memory locations are a function of the devices memory and the size of the firmware, i needed to eliminate that. 
 I could / should have asked , but the hardware was within arms reach.
( still would be interested though - and I think simple to add the the docs )

urish, 2024-01-29:
Not taken as a flame, rather as an opportunity to get some useful feedback.

1. In theory, just concatenating the two files should work. But if you can share a UF2 file where Wokwi gives you nothing (but you get feedback on the actual hardware), that will be useful. 
2. What do you mean by the memory that is assigned to virtualized devices? The flash memory? It should be the same as the hardware (for ESP32, you can actually [configure it](https://docs.wokwi.com/guides/esp32#flash-and-memory-size))

Josverl, 2024-02-06:
Sorry - no non-working files survived testing, but I remember that it was when i wanted to test with smaller and larger Flash memory sizes. 

On the memory - cool to be able to set both - IOIHK, that would have probably fixed the first issue, at least on the esp32.

callzhang, 2025-03-31:
Any updates on this? I can use Micropython on the online simulator, but needs additional configuration to run on local simulator. I am just wondering if there is a easier way to run MicroPython on my local simulator.
Thanks!

urish, 2025-04-28:
Not yet - but it's on my list to look into simplifying this.